Ultimaker/Cura #2617: Cura 2.7 and 3.03 always crash when connecting Ultimaker3 via network on macOS Sierra

soundTricker, 2017-10-18:
# Cura Version
3.0.3 (also 2.7)

# Ultimaker3 version
3.7.7.20170627

# OS
macOS Sierra (10.12.6)

# Issue 
Cura 2.7 and 3.03 always crash when connecting Ultimaker3 via network

# Steps to reproduce:
1. Open Cura
2. Settings > Printers > Add Printer
3. Select Ultimaker3 and Click "Add Printer"
4. Click "Connect" button for showing Ultimaker list.
5. Crash

# Log

Only tail lines of log file.

```
2017-10-18 11:30:41,259 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-10-18 11:30:41,261 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Cura_SteamEngine version DEV
2017-10-18 11:30:41,263 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] Copyright (C) 2017 Ultimaker
2017-10-18 11:30:41,264 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-10-18 11:30:41,266 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] This program is free software: you can redistribute it and/or modify
2017-10-18 11:30:41,268 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] it under the terms of the GNU Affero General Public License as published by
2017-10-18 11:30:41,269 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] the Free Software Foundation, either version 3 of the License, or
2017-10-18 11:30:41,271 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] (at your option) any later version.
2017-10-18 11:30:41,273 - DEBUG - UM.Backend.Backend._logSocketState [178]: Socket state changed to Connected
2017-10-18 11:30:41,275 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-10-18 11:30:41,276 - DEBUG - UM.Backend.Backend._onSocketStateChanged [168]: Backend connected on port 49674
2017-10-18 11:30:41,277 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] This program is distributed in the hope that it will be useful,
2017-10-18 11:30:41,278 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2017-10-18 11:30:41,280 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2017-10-18 11:30:41,282 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] GNU Affero General Public License for more details.
2017-10-18 11:30:41,292 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-10-18 11:30:41,295 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] You should have received a copy of the GNU Affero General Public License
2017-10-18 11:30:41,296 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2017-10-18 11:30:51,591 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-10-18 11:31:18,275 - DEBUG - UM3NetworkPrinting.DiscoverUM3Action.setKey [100]: Attempting to set the network key of the active machine to ultimakersystem-ccbdd30023af._ultimaker._tcp.local.
2017-10-18 11:31:18,276 - DEBUG - UM3NetworkPrinting.NetworkPrinterOutputDevicePlugin.reCheckConnections [222]: Connecting [ultimakersystem-ccbdd30023af._ultimaker._tcp.local.]...

```

# Note
If you need crash report via macOS, let me know.

ngineerken, 2017-10-18:
My Cura 3.0.3 crashed on first start too.

soundTricker, 2017-10-18:
Note:
I also always got crashing Cura, if I didn't delete config files for Cura at upgrading to 3.0.3 or 2.7.

Appesteijn, 2017-10-18:
Ok just tried to reproduce this, same firmware, MacOS 10.12.6, Cura 3.0.3. Printer gets added normally.
Have you tried backup/removing you settings folder? ~/Library/Application Support/cura 
Would you be willing to share your config data?

soundTricker, 2017-10-18:
@Appesteijn 
I've tried backup/removing and I got this issue. 
And I don't removing that, I always got crashing.

What config data do you need? All of  `~/Library/Application Support/cura`?

Appesteijn, 2017-10-18:
Yes please

soundTricker, 2017-10-19:
@Appesteijn [here](https://www.dropbox.com/s/n4g9paw178s9b2z/cura.zip?dl=0&m=)

soundTricker, 2017-10-19:
@Appesteijn  
Config directory is [cura.zip](https://github.com/Ultimaker/Cura/files/1396923/cura.zip)

And stderr.log is [stderr.log](https://github.com/Ultimaker/Cura/files/1396925/stderr.log)

Appesteijn, 2017-10-19:
@soundTricker Ok I just tried with your config, but I works on my Mac and I see 3 installed UM3s.
I made a bug ticket on our board so someone else can have a look.

Devs: CURA-4667

fieldOfView, 2017-10-19:
Also see #2622, which seems to have the same crash mechanism, though it happens in a different plugin.
The crash happens when using QNetworkAccessManager to get() a QNetworkRequest.

soundTricker, 2017-10-19:
@Appesteijn Thanks :)

Ghostkeeper, 2017-11-02:
We still can't reproduce this but. I'm updating it here because our internal ticket (CURA-4467) was just closed because nobody could reproduce it.

llakie, 2017-11-04:
Hi there, I got the exact same issue with Cura 2.7 and Cura 3.0.3 running it on macOS High Sierra and Sierra. It stopped working, when I upgraded to Cura 2.7. I had a look at the crash report and it's the same QNetworkAccessManager getting a new Network Request instance ans mentioned above. I have no clue what to do now.

Ghostkeeper, 2017-11-10:
Maybe we can't reproduce this bug since we're all trying to reproduce it on the same network...

bendoerr, 2017-12-09:
@Ghostkeeper I think your comment is spot on. Pretty sure I know what is going on here. I use Automatic Proxy Configuration and I've known it to cause issues with stuff before including Outlook.
No idea what the fix is, but I can offer you my situation to recreate.

You'll need a PAC file. Here is a simple one "direct.pac" that doesn't actual proxy anything.
```javascript
function FindProxyForURL(url, host) {
    return "DIRECT";
}
```

Now either though the UI. System Preferences -> Network -> Advanced -> Proxies. Enable Automatic Proxy Configuration and set "file://\<path>/direct.pac". Or run the following command

```
networksetup -setautoproxyurl "Wi-Fi" "file://$(pwd)/direct.pac"
```

Now Start Cura. If you've already got Octoprint setup, it should crash before the UI comes up. If not go set some Octoprint network settings. You could also serve it over http as well.

Now you can unset the PAC file and Cura starts again.

```
networksetup -setautoproxystate "Wi-Fi" off
```

Anyway, that is what it was for me. Even enabling the PAC while Cura is running will instantly crash it if Octoprint settings had been previously configured and connected.

ChrisTerBeke, 2017-12-10:
@nallath this thread might be interesting for your refactor?

nallath, 2017-12-15:
We're still using the same qnetwork manager, so i won't touch this now. It's big enough as it is without adding all kinds of proxy magic to the equation.

leliep, 2017-12-16:
Same for me. Always when I connect to the network via my original Apple USB Gigabit adapter A1433 EMC2590. Cura 2.8, 3.0, 3.1 each crash immediately when trying to connect to the UM3.

The fast ethernet adapter as well as Wifi are working flawlessly.

soundTricker, 2017-12-19:
I also use proxy (PAC file), then i remove this setting, Cura does not crush.

bendoerr, 2017-12-19:
Possibly related QT issue https://bugreports.qt.io/browse/QTBUG-56747

ChrisTerBeke, 2017-12-19:
That ticket also indicates a fix in 5.9. We'll probably switch to 5.10 (just released as stable) once PyQT is up to date with that as well.

bendoerr, 2017-12-19:
Excellent! If you want to update this issue when you've released with an updated PyQT, I'll test right away.

leliep, 2017-12-20:
For me it also was the proxy setting, not the ethernet interface!

ianpaschal, 2018-03-09:
Since PyQt 5.10 still caused a lot of bugs we tried to upgrade and then rolled back. Since there's no fix in sight, I'm marking this as `Deferred`.

bendoerr, 2018-03-09:
That is a bummer. Deferred until ...?

fieldOfView, 2018-03-09:
Is going to Qt 5.9 an option? That should also fix the network issue.

Appesteijn, 2018-03-09:
We do want to update to 5.10 (or higher) but currently have other things we need to finish. After that we will retry the upgrade to a higher Qt&PyQt version.

ianpaschal, 2018-03-09:
@fieldOfView Nope. Also full of bugs apparently (I don't know what, but when I joined Ultimaker in January and was setting up my dev environment it was brought to my attention that we use 5.8.2 because 5.9 causes more problems than it solved).

As for the reason for 5.10 deferment, one of the most glaring issues is the lack of system styling (or something related to that) that makes all secondary windows and pop-ups look like... actually even worse than... Windows 95.

The likely solution is that we restyle "ALL OF THE THINGS" and as @Appesteijn said, we don't have time for that in the foreseeable future.

fieldOfView, 2018-03-09:
In the mean time, perhaps @awhiemstra can shed some light on applying a patch to the current Qt 5.8 for OSX to fix the issue locally?

Ghostkeeper, 2018-03-28:
5.9.2 caused a major bug but 5.9.3 was fine afaik.

mkearl, 2018-04-04:
I get an exit crash as well since 3.2.1.   I would get this as soon as I load Cura and there was no way for it to load.   I removed the version completely with all directories etc from library/application support etc so system was clean.  I reloaded 3.2.1 and discovered that my OctoPrint feature was no long in build but needed to be installed via plugin etc.  I installed and everything was working great until I actually tried to put in the api code for OctoPrint.  If I pasted or even typed in one number Cura crashes.    I have since tried the new 3.3 beta and it acts exactly the same way.  Going to have to downgrade to 3.1 in order to even use it.   I think 3.2.1 was crashing immediately when I first installed it because it was trying to connect to OctoPrint from the last version that I had it installed with.  So frustrating!   Please fix.  Once you use Octoprint you can't just go back to sneaker net and SD cards.....

Max123459, 2018-06-05:
I have the same issue, permantly trying to check API Key. (High Sierra)

LipuFei, 2018-06-12:
Hi all, we have created a 3.4 BETA build for mac with Qt 5.10 hoping it can fix this issue. Could you try it out see if problem still exists? Thank you all!

Download link: https://we.tl/43i1hChi8m
In case the link expires, let us know so we can upload a new one.

LipuFei, 2018-06-12:
Sorry guys... I only uploaded the linux AppImage, this is the link which contains all installers/AppImage/DMG.
https://we.tl/MLmTYeGuW5

Max123459, 2018-06-12:
Can not confirm working.

Plugin Browser does not work (only installed plugins) and the "old" OctoPrintPlugin.curaprofile also does not work.

LipuFei, 2018-06-12:
The "Plugin Browser" is now called "Toolbox", and indeed that the "old" OctoPrintPlugin won't work with 3.4, but you can install it via the "Toolbox". Could you try that? And could you check if the initial issue that adding a network printer for UM3 still crashes? Thanks

Max123459, 2018-06-12:
> is now called "Toolbox"
Thats what I've meant. Sorry!

> but you can install it via the "Toolbox"
Yes, I've tried that. But there's an error: Could not connect to the Cura Package Database. Please check your connection.

Max123459, 2018-06-12:
> is now called "Toolbox"
That's what I've meant.

>  but you can install it via the "Toolbox".
Yeah, last time I got a error. After a few minutes I realized that it's caused by my Network Blocker and I had to enable access for Cura.

LipuFei, 2018-06-12:
Hi @Max123459 , sorry that the previous uploads were not correct... Could you try this one? Thanks!
https://we.tl/gu0kqipSX6

fmsarubbi, 2018-06-29:
Is there a new link that might work for osx? The link above expires. I am seeing the tool box crash Cura on 3.4.0?

LipuFei, 2018-06-29:
Hi @fmsarubbi , there is a new link to download 3.4 with Qt 5.10 for osx:
https://we.tl/RQxjmjVR2f

bendoerr, 2018-06-29:
@Ghostkeeper did you ever try my steps to reproduce by creating a simple PAC file?

Ghostkeeper, 2018-08-13:
No, I didn't since I don't have Macintosh available. Could you try in 3.4 if it's fixed for you then?

bendoerr, 2018-08-13:
Yes confirmed.