Ultimaker/Cura #2618: Cura 3.0 updater

Thisismydigitalself, 2017-10-18:
Cura 3.0 prompted me there is a new version. pressing Download did nothing. after restarting cura 3.0 and rechecking for updates Cura  prompted: "no new version was found". checked help for version number: still on 3.0.0. so - autoupdater does not work for me.

What i did next was to download 3.0 again. I should note the latest version is named: 3.0 and not 3.0.3. when download starts the file reads 3.0.3.exe. i clicked on it to install it and a message poped up:
[image]

Since I have both 2.7 and 3.0 installed side by size i did not know which cura will get uninstalled so i skipped the install.

Please guys - we are in a transition between two versions of Cura. you need to give us more accurate information what is going on under the hood. 

Thank you.

Thisismydigitalself, 2017-10-19:
Now this:
pops up after uninstalling 3.0 and installing 3.0.3.
[image]

Ellecross, 2017-10-19:
Haloo! The not working Download button is a known issue and an internal ticket has been logged for it. Unfortunately it was found too late. (It's actually fixed now, but it's not in this 3.0.3 version. :( ) 

The message that popped up is expected if you have not previously uninstalled the same version. In this case, if you had said 'yes' to that, it would uninstall your 3.0, and not the 2.7. 

I can say this because I never uninstall older Cura versions (I have 2.5, 2.6.1, 2.6.2, 2.7, 3.0.0-Beta, 3.0.1-Beta). In my experience, when I'm installing a newer version, Windows will always show me two messages: "Cura 2.5 is already installed" -> for 2.5 and "Cura is already installed.. do you want to uninstall." When I say yes to uninstall the second prompt, it will uninstall my 3.0 and never my older versions. 

That error message you got about the "error opening file Arcus.pyd" I'm not entirely sure how or why you got that. Perhaps that file was open and being used at the time when you uninstalled / installed? Maybe you can try uninstalling 3.0 again and make sure that there is no Cura 3.0 in \Program Files, and then install Cura 3.0 again. 

I hope this helps - I mention all these from what I experienced in the past!

Thisismydigitalself, 2017-10-19:
Thank you for your insight.
I will spend some more time on this later today and report back.

cheers.

Thisismydigitalself, 2017-10-21:
managed to install 3.0.3. 
Still using 2.7 since importing 2.7 profiles / projects from 2.7 to 3.x is not working as i would have hope and how it should work (there is a 3.0 profile bug case open).

Ghostkeeper, 2017-10-30:
For posterity if someone comes a-looking for a solution to the same problem, how did you manage to install 3.0.3?

closed 2017-10-30 by Ghostkeeper