Ultimaker/Cura #2622: Cura crashes when setting API key for octoprint

Erin-Cooper, 2017-10-18:
When attempting to input the octoprint API key Cura immediately crashes.
This issue only persists for me with Cura versions 2.7-3.0.3.
The issue also persists on both Mac OS Sierra and High Sierra

ChrisTerBeke, 2017-10-18:
Can you add the full logs after the crash? Then we can see the culprit :)

fieldOfView, 2017-10-18:
It also helps to know if it happened just one time or if it is reproducable.
NB: I am the maintainer of the OctoPrintPlugin, also see https://github.com/fieldofview/OctoPrintPlugin

Erin-Cooper, 2017-10-18:
sure thing! here's the crash log (If you'd prefer a plaintext file I'll paste it, just didn't want to clutter up the thread)

[cura_2017-10-18-153636_colin-3.crash.zip](https://github.com/Ultimaker/Cura/files/1395013/cura_2017-10-18-153636_colin-3.crash.zip)

The crash happens every single time any key is pressed while selecting the API key dialogue box with all versions of Cura after 2.6v

fieldOfView, 2017-10-18:
Unfortunately that's not a log that is very helpful to me. Can you share your cura.log? It is found in the Configuration Folder.

Erin-Cooper, 2017-10-18:
Sorry about that, here's the cura.log file.
[cura.log](https://github.com/Ultimaker/Cura/files/1395099/cura.log)

fieldOfView, 2017-10-18:
One more request... The application crashes in a way that is not logged in the normal Cura.log. There are two files that may have some more info:
~/Library/Logs/cura/stderr.log
~/Library/Logs/cura/stdout.log

Please make cura crash, and then get those files (they get overwritten when you restart Cura).

Erin-Cooper, 2017-10-18:
Here's the first file, I was unable to upload stdout.log as it was empty, I hope this isn't an issue?

[stderr.log](https://github.com/Ultimaker/Cura/files/1395208/stderr.log)

fieldOfView, 2017-10-18:
Thanks, that is helpful. The stdout.log being empty is also good to know.

The stderr.log shows me where the error happens, but not yet what to do about it. I will see if I can get the issue reproduced.

fieldOfView, 2017-10-19:
This seems to have the same crash mechanism as https://github.com/Ultimaker/Cura/issues/2617. The crash happens when using QNetworkAccessManager to get() a QNetworkRequest.

The OctoPrint plugin uses very similar code to the UM3 network printing plugin because the OctoPrint plugin started out as a fork of the UM3 network printing plugin.

Erin-Cooper, 2017-10-19:
Thanks for taking a look at it!
I just reset my system totally and tried it. It doesn't crash but just sticks on checking the API key forever, it also does this on my friends mac (newer model)
I'll see if it has the issue when booting into windows.

tigert, 2017-10-24:
Just for information, I have the same issue happening on 2.7 and 3.

fieldOfView, 2017-11-02:
Note that #2617, which I am convinced is the same underlying issue, was just closed because noone can reproduce the issue.

craigw76, 2017-11-19:
Seems like I'm experiencing the same issue as Erin-Cooper, but on Linux.

Using Cura-3.0.4.AppImage. Running Fedora 26 on two machines (one on KDE, the other Gnome) when I attempt to enter API Key. As soon as any character is entered into the API Key field, I see all the lines that begin with qt.network.ssl: messaged in the cura-console.log file.

I have never ran Cura on either workstation. I run the AppImage from command line. On first run I  accept the agreement, then added Creality CR-10 as printer. Then enter Preferences > Configure Cura > Printers. Select CR-10, then click Connect OctoPrint. Select octopi._octoprint._tcp.local. 

I get the same output on the console on both systems. I have tried manually adding the OctoPrint instance both by name and IP, and HTTP/HTTPS. Same results either way

[cura.log](https://github.com/Ultimaker/Cura/files/1485317/cura.log)
[cura-console.log](https://github.com/Ultimaker/Cura/files/1485318/cura-console.log)
[cura-stderr.log](https://github.com/Ultimaker/Cura/files/1485319/cura-stderr.log)

fieldOfView, 2017-11-19:
@awhiemstra, could it be that there’s some ssl bits that have not been packaged well?

kkalbaugh, 2017-11-29:
I'm having the exact same problem as @Erin-Cooper.   Sierra 10.12.6.   I removed the app and all the settings and started over without setting up anything.  Still the same issue.  Works fine on Windows but not my Mac.

[cura.log](https://github.com/Ultimaker/Cura/files/1512438/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/1512435/stderr.log)
stdout.log is empty

bendoerr, 2017-12-09:
Try disabling any System proxy configuration that you can. I just commented on #2617 with a way to reproduce on and off.

fieldOfView, 2017-12-09:
@kkalbaugh, @Erin-Cooper, could you confirm you are using a proxy?

kkalbaugh, 2017-12-09:
I am not using a proxy.  Just a small home network.

bendoerr, 2017-12-10:
Bummer, nothing else out of the ordinary in terms of network setup or possible interference such as McAfee, ZScaler, etc? I was getting this exact crash error until I disabled my system Proxy Auto Configuration/PAC file.

@fieldOfView Would you like me to open a separate issue besides #2617 detailing the proxy issue?

fieldOfView, 2017-12-10:
Let’s wait it out a little. There are so many issues already.

kezmdor, 2017-12-12:
I have the same issue on a clean install of Cura 3.1 on Windows 10. As soon as I just want to type one character of the API key Cura crashes. It happens all the time, there is no chance to get Octoprint work with Cura right now.

schneehorn, 2017-12-19:
I can confirm: I am using Automatic Proxy Configuration. With PAC turned on, Cura crashes when I try to connect to Octoprint (i.e. enter API key). With PAC turned off, it works just fine. Even if I disable PAC, enter the API key and connect to Octoprint, it immediately crashes when I re-activate PAC. This applies to Cura 2.7, 3.0.4 and 3.1 on OSX (HighSierre 10.13.2). So there definitely is a Proxy issue.

tigert, 2017-12-19:
I don't have a proxy as far as I know. And I get a crash as soon as I try to enter the first character. This is on the AppImage.

fieldOfView, 2017-12-19:
@tigert, that is new information. Sofar this has been identified as an OSX High Sierra issue. What OS are you on?

craigw76, 2017-12-22:
@fieldOfView I concur with @schneehorn's observation. I'm able to replicate on OSX High Sierra, works fine until I enable Automatic Proxy Discovery. Cura crashes on start. Disable PAC and Cura runs fine.

On my Fedora 26 machine using the appimage, Cura crashes immediately when typing any character in the API key field. I have no proxy configured on my Fedora machines. Auto Discovery is off. I run Cura from the command line and get all those `qt.network.ssl: QSslSocket: cannot resolve `... on the console right after pressing the key, then crash. 

I ran Cura from the terminal on OSX, and I did not see those qt.network.ssl messages there, nor in the Apple crash reporting tool, when it crashes. I guess the OSX executable works a little different than the Linux appimage in that respect.The only thing on the console is `Segmentation fault: 11
`
Attached is the info from the Apple Crash reporter
[OSX HS Cura crash.txt](https://github.com/Ultimaker/Cura/files/1581458/OSX.HS.Cura.crash.txt)

bendoerr, 2017-12-22:
The OSX issue is a QTNetwork issue as I suggested in #2617, I would bet that the linux issue is as well. Maybe this one https://bugreports.qt.io/browse/QTCREATORBUG-18137? Probably not picking up OpenSSL or the wrong version.

Ghostkeeper, 2018-01-15:
If it is that Qtnetwork bug, then setting `LD_LIBRARY_PATH=/usr/lib/openssl-1.0/` with the call of Cura could be a workaround? And upgrading to Qt 5.9 could fix the bug for real. I understand that such an upgrade (to 5.10) could be planned for the 3.3 release.

Kernelhacker, 2018-02-08:
I can reproduce the crash on Windows 10. Additionally cura crashes, when saving a sliced file, when "Send (anonymous) print information" is enabled. I would be glad to help!

ChrisTerBeke, 2018-02-08:
We haven't included Qt 5.9 or 5.10 yet in version 3.2, so that's expected. For the crash please create a new ticket so we can track issues properly. We probably also received the crash report already then.

Kernelhacker, 2018-02-08:
I created a new issue - but: If it is a qt problem - if i install the Qt libraries, how do i tell cura to use them? Something like "QT_LIBRARY_PATH="?

Ghostkeeper, 2018-03-02:
I think you need `QT_PLUGIN_PATH`. See an example here of some things you could try: https://github.com/Ultimaker/cura-build/blob/0e0d4623ca377be3f7e86b714032d9b8e13c7dfd/packaging/cura.sh#L7

LipuFei, 2018-06-12:
Sorry guys... I only uploaded the linux AppImage, this is the link which contains all installers/AppImage/DMG.
https://we.tl/MLmTYeGuW5