Ultimaker/Cura #2625: Small issues (M0 Sxx command?)

sim31r, 2017-10-19:
My "End Gcode", Marlin code. In the Repeter Host and Cura 15 work good. In the Cura 2.7, Cura 3.0 Something strange, I do not understand what. It seems to me that the program command M0 knocks down something in the command flow. A sequence of commands or dropping some of the commands from the thread.

G91
G1 Y5 X-5 E-5 F9000 ; go out
G1 Z+1
G90                         ;absolute positioning
G1 F9000 Y150 X150 ; go to center
M104 S0                     ;extruder heater off
M140 S0                     ;heated bed heater off (if you have it)

M300 S100 P100
M300 S300 P100 ;beep
M300 S600 P100 ;beep
M300 S900 P100 ;beep
M300 S1200 P100 ;beep

M106 S255 ; cooler on
M0 S30;
M106 S192 ; cooler test activity
M0 S30;
M106 S255 ; 
M0 S30;
M106 S192 ; cooler test activity
M0 S30;

M106 S1 ; cooler off
G1 X5 Y190 ; it is not work on cura!!? The interaction with the printer is lost. Errors are not displayed.
G1 X3 Y190
M0 S1;
M84 ;steppers off -this code works I do not understand why

maukcc, 2017-10-23:
G codes are buffered, M codes are not.
That means that while the G code for movement is still waiting to be executed, the M code is being executed instantly.
This is probably the reason to put that M0 code in there in the 1st place: wait a sec before you execute M84
If you want to wait 1sec, it is probably better to use the G4 S1 command

Ghostkeeper, 2017-10-31:
Is this about... printing via USB cable?

ianpaschal, 2018-03-09:
Needs info but did not get it for 6+ months so closing.

closed 2018-03-09 by ianpaschal