Ultimaker/Cura #2626: My plugin for Cura 15, how to transfer for the 3.0 version

sim31r, 2017-10-19:
A simple plugin for displaying the printing process on the printer screen. 

Python file:
https://www.dropbox.com/s/f2httehy0kzabof/Numerator.py?dl=0

Looked at examples of plug-ins for version 3, everything is more complicated there... Without a plug-in on the screen, you can not see the printing process:

http://3dtoday.ru/upload/main/295/2956eea26b8ff5f30cba32564d30b7fd.jpg

nallath, 2017-10-19:
You could do this with a post-processing script. 

I think it's a bit weird that the firmware of your machine does not already do this. It makes way more sense to have it there.

sim31r, 2017-10-19:
I did this in the post-processing script for Cura 15.04.6, it's very simple. For versions 2.7 and 3, the scenarios in the examples are very different and difficult to understand (for example, for me).

The program Repeater Host always sends a command to the printer this kind, it is very convenient
M117 ETE/ 1h 05m 50s*16
Time until the end of work. Every 5 second.

Ghostkeeper, 2017-10-29:
Keep in mind that what was called a "plug-in" in 15.04 is now called a "post-processing script" in Cura 2 and 3.

If you take a look at our current post-processing scripts, you'll find that they have a similar set-up as your current script: https://github.com/nallath/PostProcessingPlugin/tree/master/scripts

nickthetait, 2017-11-08:
Did that answer your question @sim31r ?

closed 2017-11-14 by ChrisTerBeke