Ultimaker/Cura #2627: Bed/Nozzle Temperature: Suggestion

Thisismydigitalself, 2017-10-19:
Can i suggest adding to Cura the ability to offset Bed and Nozzle temperatures based on real-life measurements? it will be much easier for those like me that uses open source based printers running for exp. Marlin code which in that case is somewhat cumbersome to change that in the code (I have no idea why they made it so difficult). might be due to the non-linear behavior of the temperature throughout the graph curve.

Ghostkeeper, 2017-10-29:
Can you elaborate that a little? It sounds like you're measuring a temperature 5 degrees too low and want to set the temperature to 5 degrees higher than the default, but that's what the settings are for.

You didn't intend to change the temperatures during the print based on measurements, did you? Changing the temperature too often/quickly during a print causes the PID regulator to go haywire.

ianpaschal, 2018-03-09:
Needs clarification and did not get it.

closed 2018-03-09 by ianpaschal