Ultimaker/Cura #2628: Windows HiDPI issues on second launch (Cura 3.0.3)

badvision, 2017-10-19:
I am on Windows 10, Fall Creator's Update and have a 4k monitor, set to 200% scaling

The first launch of Cura everything looks fine and the app is 100% usable.

Any subsequent launch, it scales the UI elements and fonts so large that the app becomes unusable.  I can uninstall, reinstall and then the first launch works.  But again the app becomes unusable when I launch a second time.

[image]

fieldOfView, 2017-10-19:
That is strange. Judging by the menus also being too large (File, Edit, ....), it looks like an issue with the GUI Toolkit we use (Qt). What GPU do you have, and do you have multiple monitors (at different resolutions) or just one?

badvision, 2017-10-19:
First of all, thanks for responding so quickly!  As a fellow open source developer, I have a lot of respect for that.  As for your questions:
Yes, I have a laptop (Surface Book, first generation) which has an Intel 520 integrated graphics driver as well as the performance dock with nVidia graphics support.
The icons show just fine on first start.

Now that you mention it, I tried the compatibility settings and HiDPI settings but didn't check which GPU was being used.  It was defaulting to the Intel one.  I switched to nVidia and it seemed to open fine.  So it would appear to be related to the Intel driver.  After running the latest fall creator's update I am not able to install Intel's own driver (which is a shame because it fixes an annoying HDMI sound issue that is present in the Microsoft driver -- not your problem. ;) )

You can keep the issue open if it is helpful, but thanks to your helpful question I was able to find something that works for now (use nVidia, not integrated Intel driver)

fieldOfView, 2017-10-19:
There have been some annoying issues with Intel graphics drivers lately (eg #2068). For the record, what driver version do you have for the Intel graphics?

badvision, 2017-10-19:
Windows provides 20.19.15.4463 as of now, but I can't install the Intel-supplied ones.  It would be most ideal if Intel would release a new version compatible with the fall creator's update that just released, because 154605.4771 says it is not compatible with this OS.  I was using that version before running the windows update and it was great, more or less.

fieldOfView, 2017-10-19:
@Perhaps this can help to install a newer version of the Intel GPU driver:
https://www.intel.com/content/www/us/en/support/articles/000005629/graphics-drivers.html

badvision, 2017-10-19:
It did, but only indirectly.  I had to download the ZIP version and extract to a folder, then when updating the driver I had to use the "have disk" option otherwise it would not let me select the preferred driver in the install files.  After doing that: SUCCESS!  So I think it's safe to say not a Cura problem.  But all the same, thanks for helping me solve it!

closed 2017-10-19 by badvision