Ultimaker/Cura #2629: Layer Height Markings on Recomended Settings ?

wlleven, 2017-10-19:
Love Ver. 3.0.3, but on the recommended setting tab, could you put 1/2 markings .. as in between 1.0 and 1.5 - How about a 1.2 ? I know this is a bit knit picky but I think many folks would use it.

Your Version 3.0.3 is the best yet by far, I removed Ver. 2.7 and even thinking of removing my loved 15.04.6 .. that version just works !

Thank you for the fine work you are doing for the Maker community, it is much appreciated !

wlleven

omache55, 2017-10-21:
layer height settings of v.3 got wrong one of 6 levels of layer height-it should be 0.06; 0.10; 0.15; 0.20; 0.40;
and 0.60

wlleven, 2017-10-21:
Yes, that is fine, was looking for a in between setting of .12 ... between .1 and .15

Just curious if anyone else has asked ?

wlleven

ChrisTerBeke, 2017-10-21:
The layer heights in recommended more are actually based on the quality profiles that are available for the selected printer + variant (print core/nozzle) + material. These default profiles are created and tested carefully by our materials theme to ensure high quality print results. As making these profiles takes quite a bit of time I don't see completely new profiles coming any time soon.

As @omache55 points out there are a few cases where the layer heights are incorrect due to wrong settings in the config files, that will be fixed in a next release.

Ghostkeeper, 2017-10-30:
It's a quality slider, not just a layer height slider. When you change the slider, many more settings will change than just the layer height, but we identify the quality levels by their layer height. How many settings are modified varies greatly from printer to printer.

So if we were to add a 0.12mm tick then we would either need to add a profile for that or sort of to interpolate between two profiles, but the former is a lot of work and the latter will be fairly weird because the flow rates are all wrong and such. It'd be worse than the 0.15mm profile and printing slower, that sort of thing.

closed 2017-10-31 by ChrisTerBeke