Ultimaker/Cura #2630: Cura 3 Mac OSX UI Glitch

Intoxinator, 2017-10-19:
The print bed appears completely black when starting Cura on my Mac Mini. Models load fine, slicing works and have had successful prints. The print bed just does not appear with the grid.

Screenshot here: https://www.dropbox.com/s/f2o5hg8kiv70oi0/Screenshot%202017-10-19%2023.17.24.png?dl=0

Intoxinator, 2017-10-19:
Cura 3.0.3
Mac OS 10.13
Mac Mini (Mid 2011) 
Intel HD Graphics 3000 512 MB

ChrisTerBeke, 2017-10-20:
We've seen this happening with certain graphics cards. A fix was already made but not added to the 3.0 release. Not sure if there's gonna be an updated 3.0 release that contains a fix for this, otherwise it'll be fixed in 3.1.

closed 2017-10-20 by ChrisTerBeke

foreachthing, 2017-10-20:
Same with Windows 7 Pro and Quadro FX 3800 (341.92).

fieldOfView, 2017-10-20:
Does it make sense to cherry-pick the fix to the 3.0 branch already, just in case there is a Cura 3.0.4? It would be a shame if it was overlooked again.

fieldOfView, 2017-10-23:
It was just confirmed that this is more than just an aesthetic issue; on some gpu/driver combinations it causes a crash, see #2644