Ultimaker/Cura #2631: Cura3 MacOs X - 1st layer temperature not catched

nounours2099, 2017-10-19:
Hi.

I set a 1st layer temperature, and the GCode do not contains this temperature. I set 210° as 1st layer, and 200° as printing temperature.
Gcode content :
`M104 S200 ; set extruder temp
M140 S55 ; set bed temp
M190 S55 ; wait for bed temp
M109 S200 ; wait for extruder temp`

Should have been:
`M104 **S210** ; set extruder temp
M140 S55 ; set bed temp
M190 S55 ; wait for bed temp
M109 **S210** ; wait for extruder temp`

Cura 3.0.3
OS - MacOs Sierra (10.12.6)
iMac Late 2009

Regards.
Arnaud

ChrisTerBeke, 2017-10-20:
Mm I cannot reproduce this. I'm getting `M109 S245` after I've set the initial layer temperature to 245 but the normal printing temperature to 240. I'm also not seeing the comments behind every line (`;set extruder temp`). Which G-code settings are you using?

nounours2099, 2017-10-20:
Please find below my settings:

[http://arnaud.christel.free.fr/Arnaud/Capture.PNG](url)

And the GCODE
M104 S200 ; set extruder temp
M140 S55 ; set bed temp
M190 S55 ; wait for bed temp
M109 S200 ; wait for extruder temp

I was on my Windows machine, and I have exactly the same issue.

closed 2017-10-20 by nounours2099

Ghostkeeper, 2017-10-30:
Thanks for the debugging. I've made a correction to be added to the next release: https://github.com/Ultimaker/Cura/commit/93e746b1b3fcbe7f8c12e17908a475b86e67ae14

commit 93e746b1 (2017-10-30): Warm up to layer_0 temperatures in start g-code

Thanks to nounours2099 for the correction.

Fixes #2631.