Ultimaker/Cura #2633: Cura 3.0 Crashes on File open

ajtomany, 2017-10-19:
Cura 3.0, 2.7 and 2.6 all crash instantly when attempting to open a file.  I've uninstalled and reinstalled with no change.  I've even attempted to do a clean install on a new machine, the same result.  All have installed in a Windows 10 64 bit operating systems.  Please help!

fieldOfView, 2017-10-19:
C:\Users\[you]\AppData\Local\cura should contain the following files:
stdout.log and stderr.log
Please zip those up, and attach the zip here.

ajtomany, 2017-10-19:
Here are the two files stderr and stdout.log in the requested zip file.

-- 
Armand Tomany
Educational Shop Manager / Instructor
ETAS Bldg, Room 329
University of Arkansas at Little Rock
2801 South University Ave.
Little Rock AR 72204-1000
Phone: 501-569-8025
Fax: 501-569-8039

facebook.com/ualittlerock | twitter.com/ualr

[image: UA Little rock Logo Signature for Emails]

ajtomany, 2017-10-19:
[stderr and stdout log.zip](https://github.com/Ultimaker/Cura/files/1399541/stderr.and.stdout.log.zip)

ajtomany, 2017-10-19:
[image]

Ellecross, 2017-10-26:
Hello @ajtomany, you seem to have the same issue as #2644 

I see you have Google Drive installed in your machine. As I've mentioned in the other ticket, we saw this problem before with another user and apparently it was caused by either Google Drive or Adobe  Extensions.  

Can you try removing them just to see if it will resolve your issue? 

That user uninstalled either Google Drive or an Adobe extension (not sure which one caused it, but he uninstalled both) and after that, it worked for him.

Ghostkeeper, 2017-11-05:
We consider removing the Google Drive extension a workaround, but there's not much we can do about this since it's the library we're using (Qt) that has this bug.

closed 2018-03-09 by ianpaschal