Ultimaker/Cura #2634: Deb package vs Appimage

ljoets, 2017-10-19:
Hello,
I need to install Cura3 on every computers (in a school) using Debian.
Appimage is not very simple to use : no standard shortcuts, no icon...

Is it possible and is there a documentation / howto to make a deb package of Cura3 ?
Thanks.

nallath, 2017-10-23:
You could try the PPA made by Thopiekar. We stopped creating *.deb packages as it caused us a load of issues.

ljoets, 2017-10-24:
How can I manage the installation of icon and launcher in Debian using an Appimage ?
Is there a doc to compile it under Debian Stretch ?
Thanks

sbts, 2017-10-29:
@nallath While I know it's not that hard to do a google search for Thipiekar's PPA, it really helps avoid confusion and lost time if you are able to provide a link in issue reports.

[Cura PPA by Thopiekar](https://launchpad.net/~thopiekar/+archive/ubuntu/cura)

Ghostkeeper, 2017-11-06:
Alternatively you can create a .desktop image to the Appimage as well, such as this (not tested):

    [Desktop Entry]
    Type=Application
    Name=Cura
    Comment=Slicer software
    Icon=~/Applications/Cura/cura.png
    Exec=~/Applications/Cura/Cura.appimage

You can get the icon here: https://github.com/Ultimaker/Cura/blob/master/icons/cura-128.png

This would create a launcher with an icon that you can put on the side bar, etc.

maukcc, 2017-11-07:
Or install the main menu app:

> sudo apt install alacarte

Then open main menu app and add the appimage as new item
Make sure the appimage is excecutable.
In the main menu app you can also click on the image to insert a new one

probonopd, 2017-11-08:
> Alternatively you can create a .desktop file pointing to the Appimage

Also check out the optional `appimaged` daemon which does this automagically, and much more.
https://github.com/AppImage/AppImageKit#appimaged-usage

peteruithoven, 2017-12-14:
@probonopd I was able to install appimaged and after installation it did automatically add the .desktop file, but I still can't open a .stl directly with Cura. It's not listed in the `Open with` > `Other Application` > `Select Application`. Any idea what could be going wrong? 
Have others successfully installed the AppImage and are able to open .stl files with Cura?

probonopd, 2017-12-14:
Which operating system and version are you using @peteruithoven?

> Side note, why isn't appimaged (and AppImageUpdate) mentioned on appimage.org?

Because no one has sent a Pull Request yet...

peteruithoven, 2017-12-14:
Apologies, I'm running Elementry OS Loki (based on Ubuntu 16.04).

I did find [cura.sharedmimeinfo](https://github.com/Ultimaker/Cura/blob/master/cura.sharedmimeinfo), which is [used in the cmake file](https://github.com/Ultimaker/Cura/blob/master/CMakeLists.txt#L67-L69). 

>> Side note, why isn't appimaged (and AppImageUpdate) mentioned on appimage.org?

> Because no one has sent a Pull Request yet...

Whaha, fair point...

closed 2018-03-02 by ChrisTerBeke

Ghostkeeper, 2018-03-28:
I've added a Freedesktop-format MIME type file for Linux now to the build: https://github.com/Ultimaker/cura-build/commit/91743f17c58496bad93930265800b3ffa85c62f0