Ultimaker/Cura #2636: Feature request: x/y position selected with flag or bullet point in view

Dim3nsioneer, 2017-10-19:
There are quite a number of Cura settings which work with x/y coordinates. I find it very difficult to place those properly if they are not one of the corners.
I there fore suggest to improve the UX with flag, needle, bullet point whatever displayed in the various views (object, x-ray, layers) which the user can move there.

Ghostkeeper, 2017-10-30:
I've thought about this sometimes as well. It basically needs us to combine those sorts of settings into one 2-dimensional vector and then to "set" the vector you would click a button that activates some sort of tool that allows you to place the flag in the build volume.

However we'd also like to keep the precision that can only be achieved by allowing the user to type the X/Y coordinates themselves too. And it won't really work with relative coordinates.

Dim3nsioneer, 2018-02-05:
Did you see the idea got some interest on the forum? https://community.ultimaker.com/topic/20329-cura-seriously-needs-a-more-visible-z-seam/
We were assuming you are working on it.

Ghostkeeper, 2018-02-05:
Yeah, I did. I think that @Appesteijn didn't though or he modified his filter with which he determines an issue to be too old.

elmerohueso, 2018-07-13:
Any chance we could get this moving? It'd be great if Travels or Helpers in the Layer View would should where the Z Seams are going to be.  Maybe even an easier way to place the Z Seams location.

My work-around is to enable Coasting so I see layer gaps where the Z Seams are, then try to set them where I want them (setting them by coordinates is not easy), then disable Coasting before I print.

Ghostkeeper, 2018-08-27:
You should see some travel moves near the Z seam if you have more than 1 wall. That's how I recognise it usually, anyway.

xermalk, 2018-12-23:
Bump, messed up a 25h lithophane print just now because i forgot to reset my z-seam setting.

Please make it so the seam is clearly marked in the layer view. 

Also, as something that could be really quickly implemented and would be super helpful.
Add a marker on the build plate if the user has set specified coordinates for z seam.