Ultimaker/Cura #2637: Cura 2.6 to 3.0 upgrade - Profile issue

mpsdskd, 2017-10-19:
I'm having problems with my upgrade from 2.6 to 3.0.
After opening 3.0 only a few of my print profiles are there, most of them are missing.
I tried exporting and importing them, but this does not work either. 

After some digging around I found in the exported profiles, that do work, there are a lot of files
[image],

while in the not working profiles there is only one file.

Moving the "extruder" files, but not the main pofile file, from .local/share/cura/3.0/quality/ to somewhere else, prevents cura from starting, so I could not check, whether it's really the missing "extruder" files, which keep the profile from being shown.

Is it supposed to be the case, that on dual extrusion machines I have to have a separate config file for each extruder (even non existing ones) and that profiles from 2.6 (which worked with the same dual extrusion machine), do not work in 3.0.3?

Kubuntu 17.04, cura 3.0.3 from AppImage, cura 2.6.2 from appimage (cura 2.7 from ppa did not work).
Custom dual extrusion FDM printer

I looked though the issues concerning profiles, but found nothing quite fitting.

Pheneeny, 2017-10-20:
I'm having the same issue. I upgraded from Cura 2.7 to 3.0.3, both using the appimage on Linux Mint 18.2. From what I can tell, profiles that I created with Cura 2.5 are still there, but the profiles I created when I was using Cura 2.7 are not.

Ghostkeeper, 2017-10-30:
Yes, you're supposed to have those 8 extruder stacks. Even if you have only 2 extruders, it will create the rest of the extruders in the background and just not display them, because of some implementation limitations.

I can't reproduce your problem on my end. All of my custom profiles from 2.6 ended up fine in 3.0.

Perhaps it'd be easier to debug this if you could provide some sample files that didn't translate well to 3.0? Maybe you could .zip up that 2.6 folder.

mpsdskd, 2017-10-31:
I'm not quite sure, which ones gave problems, as I recreated them in Cura 3 manually.
But I think those two gave problems.

[quality.zip](https://github.com/Ultimaker/Cura/files/1430833/quality.zip)

Ghostkeeper, 2017-11-07:
They seem to work fine when upgrading them from 2.6: 
[image]

closed 2017-12-21 by ChrisTerBeke