Ultimaker/Cura #2640: about support mesh

651851996, 2017-10-20:
How to use support mesh，drop down support mesh and anti overhang mesh？？？

Vandrasc, 2017-10-20:
See the discussion on the forum for more information: https://ultimaker.com/en/community/35603-support-mesh-anti-overhang-mesh
I hope this helps, otherwise feel free to come back :)

651851996, 2017-10-20:
I think I can not guess this in any way XD

Vandrasc, 2017-10-21:
Hi 651851996,

See the below Word document for a small guide about the Support/Drop down/Anti overhang mesh, I hope this makes it a bit clearer on how to use them. For the Support and drop down mesh you don't need the Generate Support to be enabled. Good luck! 
Let me know if you could download the file. 

[Support:Drop down:Anti overhang mesh.docx](https://github.com/Ultimaker/Cura/files/1404157/Support.Drop.down.Anti.overhang.mesh.docx)

651851996, 2017-10-21:
Thank you so much，I already know how to use it.
（This method is really hard to find by myselfヽ(￣▽￣)ﾉ）

closed 2017-10-23 by nallath