Ultimaker/Cura #2641: error in slice

lamerhouse, 2017-10-20:
as you can see from the picture that there are inconsistencies and this is reflected on the printed object
[image]

ChrisTerBeke, 2017-10-23:
Can you attach a project file so we can debug this?

lamerhouse, 2017-10-23:
[bocchettone giusto.zip](https://github.com/Ultimaker/Cura/files/1408791/bocchettone.giusto.zip)

Ghostkeeper, 2017-11-05:
Yup, reproduces for me too (using UM3 default profiles AA0.4/PLA/Fine).

It's the overlap compensation kicking in. The hollow cone has a shell that's approximately 0.6mm thick. It wants to place two walls there (one for the inner border and one for the outer border). But two lines of 0.35mm aren't going to fit, so it makes one line of 0.35mm and another of 0.25mm.

The problem is that sometimes it decides that the inside border needs to be 0.35mm and the outer 0.25mm, and sometimes the other way around. Since the overlap compensation only adjusts flow and not the positions of the lines and you only see the outside line, you will see the 0.35mm lines as protruding farther than the 0.25mm lines.

When printing this effect will be lesser, since the material overflows where the line doesn't fit, but since the printer is imperfect you will still see the effect a bit on the actual surface.

To solve this we'd need to find a way to canonicalise which of the two overlapping lines gets reduced. I don't see a way to do that directly though...

BagelOrb, 2018-03-06:
The second printed line is always the one being compensated.

It seems that it's not always printing the outer outline first.

I think it would be a good idea to always print the most outside outer wall first.

jackha, 2018-08-21:
As of Cura 3.4.1, this seems to be fixed.

[image]

closed 2018-08-21 by jackha

lamerhouse, 2018-08-22:
perfect