Ultimaker/Cura #2642: Aborting bug

sfornengo, 2017-10-20:
On printerBot simple, if I click abort when printing, the process stop and then the nozzle leave to its home position. That is what is expected. But the problem is it doesn't stay here, it then immediately return in the last printed position and then go straight to Z=0, crashing the nozzle in the printed part !

nallath, 2017-10-23:
This seems to be a bug specific to the printerBot simple. This is not something we will pick up as we don't have any printerBots. If anyone in the community wants to pick this up; pull requests are always welcome.

clint-journaltech, 2017-10-29:
This occurs on a Hictop 3DP-11-ATL too.. it didn't happen in the previous version. I've had to roll back.

Ghostkeeper, 2017-11-06:
This is probably caused by this change: https://github.com/Ultimaker/Cura/commit/6d2b46244f910cc5f09b07bdde1955bdcde77afb

The change was partially reverted here: https://github.com/Ultimaker/Cura/commit/371ca360470fd2f2521037fba489e20dddcd449f

The homeHead method there sends `G28 X` and `G28 Y` to the printer, which should instruct it to home the X and Y axes. See [documentation](http://reprap.org/wiki/G-code#G28:_Move_to_Origin_.28Home.29). All printers proclaiming they implement g-code should support this (except Delta printers?).

clint-journaltech, 2017-11-06:
The linked documentation also states:

> Because the behavior of G28 is unspecified, it is recommended not to automatically include G28 in your ending GCode. On a Cartesian this will result in damaging the printed object.

Ghostkeeper, 2017-11-11:
What shall we do instead then? Maybe we should move to position 0,0? Some printers have their 0,0 coordinate in the centre of the build plate though.

fieldOfView, 2017-11-11:
> What shall we do instead then?

Making it configurable seems to be the only option. I will see what I can do, PR-wise.

Ghostkeeper, 2017-11-11:
Maybe moving 5mm up can be considered safe on all printers.

clint-journaltech, 2017-11-13:
Configurable is great.. maybe default to moving 5mm up :)

closed 2023-06-24 by github-actions[bot]