Ultimaker/Cura #2643: 3.0 German Translation - Button text overlaps time information

nicolinux, 2017-10-20:
A german user reported this issue (Windows 7 64bit).
[image]

hannover345, 2017-10-20:
[image]
[image]

Vandrasc, 2017-10-21:
Devs:4478

Ghostkeeper, 2017-10-30:
Is there a shorter German text we could use than "Speichern auf Wechseldatenträger"? Or maybe we should elide that as well just to be sure...

fieldOfView, 2017-10-31:
See #2700

hannover345, 2017-10-31:
I can suggest for Speichern auf Wechseldatenträger
Speichern auf SD/USB
or
Speichern auf SD

On my last Post, second image
"Open Octoprint..."  "Pausieren"  "Drucken Abbrechen"
may be you can change with    

"Octoprint..." "Pausieren"  "Druck Abrechen"

fieldOfView, 2017-10-31:
Problem is, we don't know if it is an sd-card. Could be a Compactflash card, or a floppy disk.

I am considering changing the "Open OctoPrint..." button to a graphical button (with a monochrome OctoPrint logo) so it takes up less space.

nicolinux, 2017-10-31:
Couldn't we just use "Speichern"? That's pretty general and could by anything compactflash card, floppy, hard disk...

ChrisTerBeke, 2017-10-31:
@nicolinux we use different button texts for outputting to disk or to removable drive (to indicate you've selected the correct output method).

Ghostkeeper, 2017-11-07:
[Wikipedia](https://de.wikipedia.org/wiki/Wechseldatentr%C3%A4ger) suggest "Wechselmedium" as alternative. Is that better, German-speaking folk?

ianpaschal, 2018-02-28:
This hasn't been touched in 3+ months and is marked as Fixed in JIRA so I'm marking it the same here and closing it.

closed 2018-02-28 by ianpaschal

Ghostkeeper, 2018-03-23:
Corrected the text in question to `Speichern auf Datenträger`, which was suggested by a colleague. It's a bit better now.

commit 749846e0 (2018-03-23): Shorten 'Save to Removable Drive' translation

Wechseldatenträger was a bit too long to fit on that button.

Fixes #2643.