Ultimaker/Cura #2644: Cura crashes up to and including 3.03

jltott, 2017-10-20:
Have never been able to get this running. ALWAYS fails when you open a dialog box (save or open). Tried just about all releases on clean installs up to and including 3.03. Desperately want to try this but cannot. Windows 10 machine. Have requested help numerous times but never got any, one final time before I give up and ignore it forever...... Win 10 Home, Version 1703 OS build 15063.674 Intel I7@3.7GHz 16GB RAM

chputney, 2017-10-21:
I installed Cura 3.03 to use all the new features on my Win7 machine.  After installation,  crash on opening.  Finally uninstalled, deleted all cura appdata (why doesn't the uninstaller delete the appdata) and reinstalled.  Now it opens, but crashes when I define a printer (Ultimaker original).

Stdout is empty.

Here is stderr.log:
[stderr.log](https://github.com/Ultimaker/Cura/files/1404029/stderr.log)

Here is cura.log:
[cura.log](https://github.com/Ultimaker/Cura/files/1404030/cura.log)

Vandrasc, 2017-10-21:
Hi jltott, 
Did you try to remove the AppData Cura folder? From both Roaming as Local? 

chputney the AppData isn't automatically removed when uninstalling Cura by choice as some of the users still want to use their old configuration (printers, profiles etc.). Did you also remove the AppData from Local? 

I can't reproduce the crash :( Maybe you could share with us your user directory before removing it? It may be something in your old configuration that causes the crash. Thank you!

jltott, 2017-10-21:
Hi 
Thought I had removed everything but in checking again I found I still had 2.7 as well as 3.03 AppData folders. 
Saved them seperately, uninstalled 3.03 and deleted all appData folders referencing Cura
Reinstalled 3.03
Same problem, opened but crashed as soon as I tried to open a file
Here is stderr.log
[stderr.log](https://github.com/Ultimaker/Cura/files/1404208/stderr.log)
cura.log is empty

Only files in user directory are machine files in .......\AppData\Local\cura\3.0\cache\definitions\3.0.3 the prusa i3 MK2 I am selecting appears to be the standard 324KB

All other/old Cura info now gone

Thanks

jltott, 2017-10-21:
Sorry, wrong button

jltott, 2017-10-21:
For info, if it is likely to help, the screen shots attached show sequence of failure since 2.3(?)
[Cura screenshots.zip](https://github.com/Ultimaker/Cura/files/1404247/Cura.screenshots.zip)

jltott, 2017-10-21:
Just to add confusion..... just installed MeshMixer. Guess what, it does exactly the same.

chputney, 2017-10-22:
In trying to find a solution to Cura 3.03 crashing when adding a printer, I loaded Cura 15.04, Cura 2.4, and Cura 2.7.  Using the same Win 7 machine with all these releases, I was able to add a printer.

Tried again with Cura 3.03.  First uninstalled Cura 2.4, Cura 2.7 and Cura 3.03 using windows uninstaller and a.  Then deleted the  directory  appdata/Local/cura/3.0 and the directory appdata/Roaming/cura/3.0.

Reinstalled Cura 3.03 as administrator and accepted default installation options.

Cura starts with no problems and presents the "Add Printer" window.  I select "Ultimaker Original" and click Add Printer.

Cura crashes after briefly flashing the next window.

Here are the logs:
[cura.log](https://github.com/Ultimaker/Cura/files/1404855/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/1404857/stderr.log)

Looking at the logs, it seems that for some reason the printer data cannot be accessed.

There is nothing special about my Win7  machine.  What could it be ?

chputney, 2017-10-22:
Sometimes, the Add Printer gives a different error.

[stderr.log](https://github.com/Ultimaker/Cura/files/1404965/stderr.log)

[stderr.log](https://github.com/Ultimaker/Cura/files/1404961/stderr.log)

fieldOfView, 2017-10-23:
I would not be surprised if this were caused by https://github.com/Ultimaker/Cura/issues/2630

The shader that should draw the grid lines is broken on older GPUs. A fix was made, though it was not included in the 3.0 release. Some drivers draw the buildplate as a black plane, but it looks like some drivers may cause Cura to crash because of the broken shader.

You can fix this by replacing C:\Program Files\Cura 3.0\resources\shaders\grid.shader with this file:
https://raw.githubusercontent.com/Ultimaker/Cura/1313e01f04838ac0a2d83b332dc57b6f5db0b87c/resources/shaders/grid.shader

chputney, 2017-10-23:
This was the solution.  I replaced the file and was able to add a printer and start slicing.

Thank You !!!

jltott, 2017-10-23:
Willing to try anything but sorry to be thick - how do I replace the file on my computer.

fieldOfView, 2017-10-23:
* Download the file (follow the link above, go to File -> Save Page As...)
* In explorer, open the folder C:\Program Files\Cura 3.0\resources\shaders
* Drop the file you downloaded into that folder
* In the dialogs that pop up, allow overwriting the file

jltott, 2017-10-23:
As I said, willing to try anything but as I thought, no solution. My problem is not one of choosing a machine but loading a file. Any other ideas anyone? By the way. Thank you anyway, I am glad it solved one persons problem.

fieldOfView, 2017-10-23:
I am sorry, my solution was specifically for a 3.0 issue only, when drawing the buildplate for the first time. I had not noticed @chputney had taken the issue off topic...

Ellecross, 2017-10-26:
@jltott Hey your problem sounds familiar. I remember this problem was seen before with another user. Do you have happen to have Google Drive or Adobe Shell extensions? 

Can you try removing them just to see if it will resolve your issue? Previously, this problem was seen happening with another user. He uninstalled either Google Drive or an Adobe extension (not sure which one caused it, but he uninstalled both) and after that, it worked for him.

jltott, 2017-10-26:
Please keep open, thanks for the suggestions, I have to go into hospital for a week so will get back to this sometime after that. Thanks for all the help

Ellecross, 2017-10-26:
Ack! I hope all goes well @jltott <3

closed 2018-03-02 by ChrisTerBeke

pegasis010, 2018-06-03:
I have this issue, no google drive or adobe other than adobe reader.
Cura 3.3.1 crashes on open or save dialog box.
Anyone else seeing this again?

jltott, 2018-06-03:
Hi, for info I never got this working, certainly was never fixed, just moved on. deleted everything suggested, even deleted all Adobe references and files. Just gave up using (trying to) cura and meshmixer.

pegasis010, 2018-06-05:
So in the off chance anyone other than jltott and myself are looking at this, the crash is happening when any windows file management dialogue box within Cura opens. Open, Save as, Export materials etc......

jltott, 2018-06-05:
And if anyone does come up with a solution I would love to hear it.

Ghostkeeper, 2018-07-18:
We haven't reproduced this issue ourselves here in the office but we've gotten reports about this from time to time. The problem back then was that some Windows extension (Adobe, Google) modified the file dialogue and this caused Qt to crash whenever Cura opened that file dialogue. You might still have some other modifications to the file dialogues, even ones that aren't very visible.

jltott, 2018-07-18:
Many thanks for that, the trouble is I don't know what has been modified or
how to change it back

pegasis010, 2018-07-18:
It may help to k ow the issue is also happening with Autodesk Meshmixer.
But only these two software. Every other software I use is fine.
SolidWorks, Inventor, MS Word, OpenScad, Slic3r etc.

jltott, 2018-07-18:
Agreed, my mesh mixer also gave up at the same time

fieldOfView, 2018-07-18:
MeshMixer also uses Qt 5.8. Perhaps this will finally be fixed now that Cura is being upgraded to QT 5.10.

jltott, 2018-07-18:
Sorry to be thick but what's qt

Ghostkeeper, 2018-08-13:
[Qt](https://www.qt.io/) is the software library that Cura uses to display its windows, buttons, etc.

jltott, 2018-08-13:
Thanks – do I need to refresh something I have or just not worry about it?
Rgards

John

From: Ghostkeeper
Sent: 13 August 2018 07:26
To: Ultimaker/Cura
Cc: jltott; Mention
Subject: Re: [Ultimaker/Cura] Cura crashes up to and including 3.03 (#2644)

Qt is the software library that Cura uses to display its windows, buttons, etc.

Ghostkeeper, 2018-10-12:
Can you try again using Cura 3.5?

jltott, 2018-10-12:
Will do

jltott, 2018-10-12:
No difference I am afraid – still dies when you try to open a file…….

jltott, 2018-10-12:
Still didnt work
stderr file as follows if it is of any interest
[stderr.log](https://github.com/Ultimaker/Cura/files/2473987/stderr.log)

Ghostkeeper, 2018-10-26:
This is suspected to be a duplicate of https://github.com/Ultimaker/Cura/issues/1537. We can't confirm that yet since you said that disabling shell extensions didn't work, but that issue report might have some tips for you.

pegasis010, 2018-10-26:
I switched to Slic3r and Repetier at home, had no issues so put those on
our school computers. Problem solved.
Ran a quick test and yes, Cura 3.5.1 is crashing on our Windows PCs and
laptops as soon as any dialog box opens.