Ultimaker/Cura #2645: own Post Processing Plugin (script) with appimage

hexxter, 2017-10-20:
Hi,
how can i use my own Post Processing Plugin (script) if i use the appimage? I can not mount the appimage  rw.

BR Hexxter

fieldOfView, 2017-10-20:
See https://github.com/nallath/PostProcessingPlugin/issues/48

closed 2017-10-30 by Ghostkeeper

jypma, 2018-02-19:
I've been trying to find out how to make a custom script ([`ChangeAtHeight.py`](https://www.thingiverse.com/thing:2077884)) available when running as appimage. I couldn't make it work with Cura 3.2.1, tried:
```
~/.local/share/cura/3.2/plugins/PostProcessingPlugin/scripts
~/.local/share/cura/3.2/scripts
```
So what's the magic location to place the `.py` file?

fieldOfView, 2018-02-19:
It should be ~/.local/share/cura/3.2/scripts

There may be an error message in the cura.log file.

jypma, 2018-02-19:
Interestingly, the code's auto-creation actually made a `scripts` folder at `~/.config/cura/3.2/scripts`. And placing the `.py` there, it is indeed picked up. Curious, since cura does store all other settings in `~/localshare`...

fieldOfView, 2018-02-19:
My bad, probably (I contributed https://github.com/nallath/PostProcessingPlugin/pull/51). The linux distribution is the only one that splits up the config and "other data" locations, so I never noticed it uses the "wrong" path.

jypma, 2018-02-19:
Yeah, I expected something like that. I'm using Arch linux, btw. Shall I create a new issue for this? It might just be a matter of documentation.

jypma, 2018-02-19:
Created https://github.com/Ultimaker/Cura/issues/3356.