Ultimaker/Cura #2647: Gui doesn't show up

jancoow, 2017-10-20:
After I updated my system, cura doesn't show its gui anymore at startup. It does show its loading image, but after that, the normal gui doesn't show up. 

**OS:** Arch Linux
**Version:** Cura Master

**Startup log:**
```
UM.Resources.__initializeStoragePaths [342]: Initializing storage paths
UM.Resources.__initializeStoragePaths [353]: Config storage path is /home/janco/.config/cura/master
UM.Resources.__initializeStoragePaths [361]: Data storage path is /home/janco/.local/share/cura/master
UM.Resources.__initializeStoragePaths [373]: Cache storage path is /home/janco/.cache/cura/master
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [119]: Trying OpenGL context 2.0...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [128]: We got at least OpenGL context 2.0: 3.0 No profile
UM.Qt.QtApplication.__init__ [80]: Detected most suitable OpenGL context version: 2.0 No profile
2017-10-20 22:40:45,847 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ConsoleLogger
2017-10-20 22:40:45,858 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /usr/bin/CuraEngine
2017-10-20 22:40:45,858 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraEngineBackend
2017-10-20 22:40:45,861 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLWriter
2017-10-20 22:40:45,865 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin STLReader
2017-10-20 22:40:45,866 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJReader
2017-10-20 22:40:45,868 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin OBJWriter
2017-10-20 22:40:45,868 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin ConsoleLogger was already loaded
2017-10-20 22:40:45,904 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UpdateChecker
2017-10-20 22:40:45,911 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SimpleView
2017-10-20 22:40:45,911 - INFO - UpdateChecker.UpdateCheckerJob.run [44]: Checking for new version of cura
2017-10-20 22:40:45,919 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ScaleTool
2017-10-20 22:40:45,921 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CameraTool
2017-10-20 22:40:45,924 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin RotateTool
2017-10-20 22:40:45,926 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SelectionTool
2017-10-20 22:40:45,928 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin TranslateTool
2017-10-20 22:40:45,930 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MirrorTool
2017-10-20 22:40:45,932 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin FileLogger
2017-10-20 22:40:45,935 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LocalFileOutputDevice
2017-10-20 22:40:45,937 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin AutoSave
2017-10-20 22:40:45,937 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin STLWriter was already loaded
2017-10-20 22:40:45,937 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin STLReader was already loaded
2017-10-20 22:40:45,937 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin OBJReader was already loaded
2017-10-20 22:40:45,937 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin OBJWriter was already loaded
2017-10-20 22:40:45,942 - WARNING - 3MFReader.<module> [10]: Could not import ThreeMFReader; libSavitar may be missing
2017-10-20 22:40:45,947 - WARNING - 3MFReader.<module> [10]: Could not import ThreeMFReader; libSavitar may be missing
2017-10-20 22:40:45,948 - ERROR - UM.PluginRegistry.loadPlugin [220]: Plugin 3MFReader did not return any objects to register
2017-10-20 22:40:45,950 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin XmlMaterialProfile
2017-10-20 22:40:45,953 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin SolidView
2017-10-20 22:40:45,988 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UM3NetworkPrinting
2017-10-20 22:40:45,992 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin CuraProfileReader
2017-10-20 22:40:45,996 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade21to22
2017-10-20 22:40:45,999 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade25to26
2017-10-20 22:40:46,001 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade26to27
2017-10-20 22:40:46,004 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin VersionUpgrade22to24
2017-10-20 22:40:46,004 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin CuraEngineBackend was already loaded
2017-10-20 22:40:46,005 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin ConsoleLogger was already loaded
2017-10-20 22:40:46,013 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin USBPrinting
2017-10-20 22:40:46,023 - WARNING - 3MFWriter.<module> [9]: Could not import ThreeMFWriter; libSavitar may be missing
2017-10-20 22:40:46,026 - WARNING - 3MFWriter.<module> [9]: Could not import ThreeMFWriter; libSavitar may be missing
2017-10-20 22:40:46,026 - ERROR - UM.PluginRegistry.loadPlugin [220]: Plugin 3MFWriter did not return any objects to register
2017-10-20 22:40:46,027 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin UpdateChecker was already loaded
2017-10-20 22:40:46,027 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin SimpleView was already loaded
2017-10-20 22:40:46,032 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin UltimakerMachineActions
2017-10-20 22:40:46,035 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LegacyProfileReader
2017-10-20 22:40:46,039 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin LayerView
2017-10-20 22:40:46,040 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin ScaleTool was already loaded
2017-10-20 22:40:46,040 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin CameraTool was already loaded
2017-10-20 22:40:46,040 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin RotateTool was already loaded
2017-10-20 22:40:46,040 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin SelectionTool was already loaded
2017-10-20 22:40:46,040 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin TranslateTool was already loaded
2017-10-20 22:40:46,040 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin MirrorTool was already loaded
2017-10-20 22:40:46,044 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin ChangeLogPlugin
2017-10-20 22:40:46,044 - WARNING - UM.PluginRegistry.loadPlugin [193]: Plugin FileLogger was already loaded
2017-10-20 22:40:46,047 - INFO - UM.PluginRegistry.loadPlugin [231]: Loaded plugin MachineSettingsAction
[... 26 lines cut ...]
2017-10-20 22:40:46,179 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] the Free Software Foundation, either version 3 of the License, or
2017-10-20 22:40:46,179 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] (at your option) any later version.
2017-10-20 22:40:46,179 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-10-20 22:40:46,179 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] This program is distributed in the hope that it will be useful,
2017-10-20 22:40:46,179 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2017-10-20 22:40:46,179 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2017-10-20 22:40:46,179 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] GNU Affero General Public License for more details.
2017-10-20 22:40:46,179 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] 
2017-10-20 22:40:46,180 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] You should have received a copy of the GNU Affero General Public License
2017-10-20 22:40:46,180 - DEBUG - UM.Backend.Backend._backendLog [93]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2017-10-20 22:40:46,246 - WARNING - UM.Settings.ContainerRegistry.load [173]: Could not determine container type for file /home/janco/.local/share/cura/master/machine_instances/Ultimaker+2.cfg, ignoring
2017-10-20 22:40:57,119 - DEBUG - UM.Settings.ContainerRegistry.load [214]: Loading data into container registry took 10.872111797332764 seconds
[... cut ...]
QQmlComponent: Component is not ready
2017-10-20 22:40:57,385 - WARNING - UM3NetworkPrinting.DiscoverUM3Action._createAdditionalComponentsView [148]: Could not create ui components for UM3.
2017-10-20 22:40:57,393 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [392]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0030574798583984375 seconds. 36 input vertices. 36 output vertices.
2017-10-20 22:41:07,752 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
```

fieldOfView, 2017-10-20:
See https://github.com/Ultimaker/Cura/issues/2596

Ghostkeeper, 2017-10-22:
Indeed, a duplicate.

closed 2017-10-22 by Ghostkeeper