Ultimaker/Cura #2648: Cura 3.0.1 beta crashes on mac (report log incl.)

ayildirim, 2017-10-20:
This happened on start while it tries to load up. I was using 2.7 before this.

Version: 3.0.1-BETA
Platform: Darwin-16.7.0-x86_64-i386-64bit
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/3.0/build/inst/bin/cura_app.py", line 81, in <module>
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/Application.py", line 321, in getInstance
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 206, in __init__
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 70, in __init__
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/Application.py", line 89, in __init__
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/Resources.py", line 65, in getPath
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/Resources.py", line 117, in getStoragePath
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/Resources.py", line 154, in getStoragePathForType
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/Resources.py", line 375, in __initializeStoragePaths
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/Resources.py", line 409, in _copyLatestDirsIfPresent
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/UM/VersionUpgradeManager.py", line 244, in copyVersionFolder
  File "/Users/ultimaker/build/env/3.0/inst/lib/python3.5/shutil.py", line 536, in move
shutil.Error: Destination path '/Users/ahmet/Library/Application Support/cura/3.0/cura' already exists

-------------------------
Current thread 0x00007fff9f5383c0 (most recent call first):
  File "/Users/ultimaker/build/3.0/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 100 in show
  File "/Users/ultimaker/build/3.0/build/inst/bin/cura_app.py", line 45 in exceptHook

closed 2017-10-20 by ayildirim

Ghostkeeper, 2017-10-30:
This shouldn't happen! It gives an error that moving a folder didn't work because the folder already exists, but we shouldn't start moving that folder if the destination (3.0 folder) already exists.

Did it get solved for you somehow, seeing as how you closed the issue? How did you solve it? Perhaps other people can use your solution as a workaround if they get the same issue.

ayildirim, 2017-11-02:
@Ghostkeeper I think I had older version of the Cura left open or stuck running. But while noticing that, I also noticed there is a Cura newer nonBeta version available. So I downloaded that nonbeta version on the official site. Since then I have not a problem.