Ultimaker/Cura #2650: Cura crashes when changing the color of filament in Materials Manager screen

GrueMaster, 2017-10-21:
System:  Ubuntu 16.04.3 LTS on Intel NUC6i5SYB w/ 16G memory and 500G SSD.
Cura version(s):  3.0.3 AppImage

When changing the filament color in the Materials Manager, Cura crashes hard.  

[Cura-bug.log](https://github.com/Ultimaker/Cura/files/1403888/Cura-bug.log)

Vandrasc, 2017-10-21:
I tried few times but I am not able to reproduce, do you mind sharing your user directory with us? It may be something in your configuration. Which printer are you using? 

You can find your user directory in Cura-> Help/Show configuration folder. You may have to zip the folder before uploading it. 

Thank you!

GrueMaster, 2017-10-22:
Well, since it is Linux, there are actually 2 folders for config data (~/.config/cura and ~/.local/share/cura).  Fortunately, I know my way around.  
[cura-settings.zip](https://github.com/Ultimaker/Cura/files/1404663/cura-settings.zip)

Vandrasc, 2017-10-23:
Hi GrueMaster,

The folder is empty :( Did you zipped and upload the original folder or the copy of it?

fieldOfView, 2017-10-23:
@Vandrasc, the zip is not empty. It contains two folders (.local and .config), which are probably hidden on your system after extracting them.

GrueMaster, 2017-10-24:
I didn't bother with the ~/.cura directory as it only has 15.04 config files and plugins.  Not sure what else you want.  Cura-3.0.3.AppImage->Help->Show Configuration Folder takes me to ~/.config/cura/3.0.  Misses half the config settings that are in ~/.local/share/cura (why split them up????).

Please let me know what else I can provide.

GrueMaster, 2017-10-25:
Here's video of the crash in motion.
[cura-3.0.3-bug.zip](https://github.com/Ultimaker/Cura/files/1413081/cura-3.0.3-bug.zip)

Ghostkeeper, 2017-11-05:
When trying this on Windows, I noticed that the colour is not updated in the GUI until a refresh is done. The GUI elements that are continuously on the screen (such as the extruder tabs) stay the old colour. Things that can be refreshed (layer view 3D, layer view panel) will only get the new colour after refreshing that (i.e. by slicing or by switching to Solid view and back). It seems like we should emit a signal there...

Ghostkeeper, 2017-11-05:
> I didn't bother with the ~/.cura directory as it only has 15.04 config files and plugins.

Yeah we store it in ~/.config, ~/.local and ~/.cache since 15.06, so that .cura folder is no longer of any use for debugging.

> (why split them up????)

Because the XDG base directory specification tells us to: https://standards.freedesktop.org/basedir-spec/basedir-spec-latest.html

Ghostkeeper, 2017-11-05:
Can't reproduce on Linux either. I changed your eSun Black material to have a beautiful purple colour, but that worked fine (apart from the update bug as I said before).

GrueMaster, 2018-01-25:
Sorry for not responding in a while.  I can reproduce this on both 3.1 and now 3.2 beta.  If you download my config files (above) and try to change the color for the Metalink Aqua Glow PLA to anything, it crashes.  The profile is .local/share/cura/3.2/2.7/materials/custom_material+%233.xml.fdm_material.  Interestingly, neither the adhesion_info nor the description data gets loaded into Cura.  I don't have that problem with any of the other custom filaments in my profiles.

From the console output:
file:///tmp/.mount_h0AqUT/usr/bin/resources/qml/Preferences/MaterialsPage.qml:64:24: Unable to assign [undefined] to QColor (repeated multiple times).  
and
file:///tmp/.mount_h0AqUT/usr/bin/resources/qml/Preferences/MaterialsPage.qml:64:24: Unable to assign [undefined] to QColor (repeats multiple times before core dump).

Seems relevant.

And I rather like that particular filament.

As to OS settings, I am using the stock Unity theme, with the latest system updates.

fieldOfView, 2018-01-28:
One thing that somehow went wrong is that there should not be a 2.7 folder inside .local/share/cura/3.2
Could you upload a new zip with only ~/.local/share/cura/3.2 in it (including that 2.7 folder)?

ianpaschal, 2018-03-09:
@LipuFei fixed in refactor?

LipuFei, 2018-03-09:
Fixed in master

closed 2018-03-09 by LipuFei