Ultimaker/Cura #2651: Platform adhesion maximum is calculated wrong

spockvv, 2017-10-21:
Hi Everybody.
I experience the problem that the maximum brim size is based on the max stl-part xy-thickness. This is a problem for printing parts that are big though have a small base area. For example a sphere. If youd print it to the maximum size, you need to print it without brim. Therefore the base area xy-width should be used for calculating the maximum.

Ghostkeeper, 2017-10-30:
This is a very difficult problem, since to know the area on the base you'd need to perform a slice. Not only do you need to make an actual slice-through of the model at half the initial layer height, but you'd also need to compute the support area on the base layer, with the dozens of features that belong to that. And then there's stuff like "Make overhang printable" and travel moves avoiding stuff and all that jazz.

Basically you need to make an entire slice. This just takes too much time because the whole purpose of that disallowed area is to give immediate feedback before slicing.

ianpaschal, 2018-03-09:
Related to:
#2528
#3365
#3371 

Same answer as in those places: We don't want to use a guess-n-check approach to slicing.

closed 2018-03-09 by ianpaschal