Ultimaker/Cura #2652: Cura 3.0.3: Black raster on black buildplate (and color issues)

7eggert, 2017-10-21:
System: Debian/Jessie, KDE, Oxygen theme.
Cura version(s): 3.0.3 AppImage

The Cura build plate does not show the 1 mm / 10 mm lines but it is completely black.

My printer is configured as Prusa I3 (Anet A8).

Also while writing this, I noticed that the drop down input elements are displayed with light-gray text on lighter-gray background.

fieldOfView, 2017-10-22:
This first issue is a duplicate of https://github.com/Ultimaker/Cura/issues/2630

Could you add a screenshot of the second issue?

Ghostkeeper, 2017-10-31:
A screenshot is often difficult with drop-downs on KDE.

We might mix up some system theming with our own theming somewhere, but as far as I know we don't theme the drop-down menus at all (apart from what they drop down from).

7eggert, 2017-11-03:
[image]
[image]

ChrisTerBeke, 2017-11-03:
Please check the newly released 3.0.4 version to see if the build plate issue was fixed. The dropdown menus aren't styled indeed.

7eggert, 2017-11-03:
Download failed in Opera 12 / no cookies, no third-party-cookies, no referrer, no script except ultimaker + Cloudfront; Used vivaldi instead (not my main browser until it supports privacy).

Blach build plate is fixed.

ChrisTerBeke, 2017-11-03:
The download is not arranged by the Cura team but by the web team. Closing since the black build plate is fixed now.

closed 2017-11-03 by ChrisTerBeke