Ultimaker/Cura #2653: Build Plate Temperature not getting value from material but from printer's profile

Muralha, 2017-10-21:
When changing Material, the Build Plate Temperature doesn't change value.
Printing Temperature changes correctly.

The Build Plate Temperature is using the printer's _Profile_ Build Plate Temperature and **not** _Material_ Build Plate Temperature.

Cura 3.0.3
MacOS 10.13

ChrisTerBeke, 2017-10-21:
Can you specify which printer and material settings you're using? Usually the material settings indeed override the printer settings, but it might be that there's a typo or something in a configuration file.

Muralha, 2017-10-21:
Any Material.
Other printer > Prusa i3
Then changed settings more accordingly to personal and Anycubic i3 Mega settings.

In the Profile:
Printing Temperature: Calculated
Printing Temperature Initial Layer: Calculated (probably = Printing Temperature)
**Build Plate Temperature: 60** 
Build Plate Temperature Initial Layer: Calculated (probably = Build Plate Temperature)

Muralha, 2017-10-21:
And Material Fan Speed is not changing the Fan Speed.

fieldOfView, 2017-10-22:
Once you've put something in the sidebar, that overrides what you have set in the material. This includes when you make a new profile. So if the profile includes a value for Material Fan Speed, the value specified in the material will not be used.

Muralha, 2017-10-22:
But shouldn't appear the Undo arrow or the  i circle, like the other settings?

fieldOfView, 2017-10-22:
No, not once a value is saved into a profile. The Undo arrow only shows for values that are not yet stored in a profile.

ChrisTerBeke, 2017-10-31:
So not actually a bug?

closed 2017-10-31 by ChrisTerBeke

Muralha, 2017-10-31:
It's a bug.

If Materials don't change bed temp. value, what's this value for??
Just information?
So, useless...

If every time anyone changes materials, has to go to the material's preferences to check the temperature... why would we need the Material's dropdown??? (not very user friendly)

Personally, I don't think temperature values should be stored in the Profiles.
Profiles values should only be quality/precision values (speed, layer, etc.), but temperature values are Material values.

Also, I think that the printer's profile should have:
- max. nozzle temp. (260ºC)
- max. bed temp. (110ºC)
- print speed range (20-100mm/s; recommended 60mm/s)
- travel speed range (100mm/s)
- layer resolution range (0.05-0.3mm)

And, material's profile could have:
- print temp. range (230-270ºC; user value 250ºC)
- bed temp. range (80-100ºC; user value 90ºC)
- print speed range (40-100mm/s; user value 50mm/s)

This way, the software could block the user from inputing wrong values, potentially damaging the printer or have a bad print and it would greatly simplify and improve the user experience.

Muralha, 2017-10-31:
I believe it's fixed with 3.0.4, but I think I'm seeing another bug on something else...

ChrisTerBeke, 2017-10-31:
What I got from the comments above is that the temperature was overwritten as user setting, otherwise it would use the highest temperature for the selected materials. We didn't change this in 3.0.4 (that only fixed some OpenGL compatibility issues). The other comment you placed is more a change in behaviour that a current bug. Changing where certain settings are stored is not really possible, as any overwritten setting will go in the user settings.

Ghostkeeper, 2017-11-07:
We want the settings that the user changed in the Custom settings menu to override all of the defaults that the material profile defines. They are only defaults after all.

It's impossible to divide all settings up into the ones that belong to the material profile, ones that belong to the printer, ones that belong to the nozzle, etc. since there are lots of settings which could fit in multiple or all of those.

In the Custom settings menu we will also give warnings when we think that the setting will likely mess up the print (yellow) or are plain impossible (red).

rally25rs, 2019-06-08:
This same "issue" still exists in Cura v4...

If you have a profile "normal" that has build plate temp 60, and a material "custom > pla" that has a build plate temp 70:

if you set cura to the custom material and the normal profile, it sets the temp to 60.

Ghostkeeper, 2019-06-10:
Indeed, that appears to be the expected behaviour, as explained above. If the quality profile says it should be 60, then that overrides the temperature set in the material profile.

fmalekpour, 2019-06-22:
Using Cura 4.1.0, this issue still exists and very annoying!

fieldOfView, 2019-06-22:
That is because the design/architecture has not changed, so this is still "by design"

pdzamore, 2019-07-12:
In fact, I had a very testy conversation with Ultimaker NA support this morning about this exact type of issue. In addition to the weird architecture, the material profiles transferred from Cura 4.1 to my Ultimaker S5 constantly become corrupted on the machine, changing the material-specific print temperature to 200°C. Ultimaker told me that they advise against creating custom materials in Cura, and that I should create a separate profile for every material instead? So why have custom material at all, if the information isn't used reliably? And why do material settings for print temperature, retraction distance and speed, as well as material cost/mm, take precedence over the profile settings, but build plate temperature does not?

Ghostkeeper, 2019-07-15:
> In addition to the weird architecture,

I assure you that any weirdness in the settings architecture is still the best way we could find to support the variety of printers and printer configurations. We're open to suggestions if you know of any specifics.

> the material profiles transferred from Cura 4.1 to my Ultimaker S5 constantly become corrupted on the machine, changing the material-specific print temperature to 200°C

If your material profiles become corrupt on the machine, that's worth creating a ticket for here on Github. We might be able to do something about it. Be sure to include the material profile in question or tell us how to reproduce it.

> Ultimaker told me that they advise against creating custom materials in Cura, and that I should create a separate profile for every material instead? So why have custom material at all, if the information isn't used reliably?

I disagree with them in this case. Custom quality profiles won't be transferred to the printer and so your printer will still use 200°C for things such as heating up for the auto bed levelling, X/Y calibration print, etc. Custom material profiles are the intended way of working if you're using a filament that is not by default in Cura.

> And why do material settings for print temperature, retraction distance and speed, as well as material cost/mm, take precedence over the profile settings, but build plate temperature does not?

They don't. Cura has no specifics on which settings can be overridden in which profile layer. The quality profile always overrides the material profile. You might be seeing that because some settings ARE overridden by the quality profile (and thus changes made to the material profile don't show up until you change the quality profile) and some aren't.

That said, we did split up the printing temperature and build plate temperature in two settings ("Default Printing Temperature" and just "Printing Temperature") so that the quality profile could provide a small delta on the value from the material profile. This is a little unusual, but we chose this because it's easier to understand for a user than to have a setting that indicates a "delta temperature" and handle it in the engine.

rally25rs, 2019-07-15:
I've been using Cura for over 3 years and had never really noticed this behavior until I started using different filaments. I think there are a couple reasons why it is really confusing;

1) it seems that the quality profiles actually change based on the selected filament (maybe?). If I select a PLA material and a "Normal" profile, I see one bed temp. If I change my material to PET or ABS, then the "Normal" quality profile's bed temp changes. I had always assumed that these quality profiles were in no way linked to the selected material... but maybe they are? It's very unclear.

2) As stated above, `"Cura has no specifics on which settings can be overridden in which profile layer. The quality profile always overrides the material profile."` however there doesn't appear to be a away to _unset_ a bed temperature in the quality profile, so why have the setting on a material at all, if it will never be used? (it will always be overridden by the quality profile, which must always have a bed temp)

nallath, 2019-07-15:
The quality profile doesn't _have_ to set the bed temperature. If it doesn't have it, the one from the material is used.

rally25rs, 2019-07-15:
@nallath maybe that's a big part of the confusion then. The default profiles _seem_ to have one set and to update a profile you just click the "Update profile with current settings" button. How do you unset things for a profile?

nallath, 2019-07-15:
The quality changes (aka: the profile that you save) live one level up from the quality profiles (who, in turn, live one level above material).

Ghostkeeper, 2019-07-16:
> it seems that the quality profiles actually change based on the selected filament (maybe?).

Yes, they are swapped out for different profiles. After all, printing on high fidelity with PLA requires different settings than printing on high fidelity with ABS. Also, some quality levels are just not available for some materials.

> The default profiles seem to have one set and to update a profile you just click the "Update profile with current settings" button.

To use this button you need to have made a custom profile, which gets put on top of the quality profile. To clarify Nallath's comment, these custom-made profiles are internally called "quality changes".

> How do you unset things for a profile?

You cannot unset things from a profile at the moment. The quality profiles are read-only. To unset settings accidentally overridden from a custom profile, you'd need to switch to the original quality profile (without the custom profile on top), see the setting value and then switch back and set the setting to the value you saw. We didn't find a big enough use case to make this any easier.

I feel like the ability to modify quality profiles as well will not make it any simpler to use Cura's profile system.

ksihota, 2019-07-21:
Interesting. I have the exact same issue but mine is because I changed my extruder temp in my custom profile so it no longer changes when I switch filaments. I never changed the bed temp so it works as expected. I didn't realize that this was a **never go back** alteration. It would certainly be nice to just leave the **Default Printing Temperature** field empty (or enter a standard variable) and have it revert back to its original behaviour.
I tried removing any info from the **Default Printing Temperature** field to see if that would clear it, but it just sets it to 0 degrees.

I assume that if I alter the profile file I will be able to make the necessary change.  It would just be nice if I didn't have to play with those files as they can really muck up Cura if you make a mistake.

If I don't want to play around with the Profile files, it appears now that I will have to go through all my settings for my custom profile and recreate it from scratch, making sure I do NOT alter the Extruder or Bed temps .
Is there an easy way to do this?
[image]

ksihota, 2019-07-21:
Figured out which file and setting to alter to fix the problem in my profile. 

Its not hard to fix but it would be easier to just provide a variable override like:
Default Printing Temperature = filament_setting
instead of having to go into the profile file and removing the line
**default_material_print_temperature = 200** (or whatever temp you applied)
under **[values]** and setting the other value:

> [values]
> brim_width = 3.0
> infill_sparse_density = 100
> material_flow = 100
> **material_print_temperature = =default_material_print_temperature**

Ghostkeeper, 2019-07-22:
There is currently no way to remove an override from an existing profile if the underlying profile was also a fixed value. There is one way that you can re-link an inheritance function (the little fx button) which solves 90% of the issue, but to be able to remove overrides goes so deep in expert territory that we don't think there would be large enough of a user base for it to clutter the interface.

Most people don't even know that a profile overrides some settings but leaves others intact. They just expect a profile to be a preset collection of settings.

ksihota, 2019-07-23:
Ghostkeeper, I am not sure if we are talking about the same thing.  Please clarify for me.

This is what I experienced and what I did to remedy the problem:
I had made a profile that I was using but when I switched to different filaments the extruder temperature would not change. It always used the **Default Printing Temperature** that I had mistakenly entered at some point. This was a hassle as it required me to manually change the temperature each time I changed filaments.
I was able to fix this problem by manually editing the cfg file for the profile. I simply removed the reference to **default_material_print_temperature** and reassign the **material_print_temperature** to **=default_material_print_temperature**.
After that change to the cfg file the extruder temperature changed each time I changed the filament.
Is this the issue you are referring to?
I am calling my original change to **Default Printing Temperature** an Override to the default profile settings. I believe that the filament temperature setting should be the default for all profiles, so it would automatically correct when the filament changes. **Printing Temperature** can be altered (and is resetable) if you need to make a temporary change.
Personally I don't think that you should be able to change **Default Printing Temperature** if you can't change it back. If your profile is based on a profile that did change this to a static temperature the ability to reset it should always be there.
If I can make the change in the cfg file I don't see why it can't be programmed in.

Does this become a problem when dealing with multiple extruders or mixed filaments? I am only running a single extruder system so I don't know what other issues arise when you have more than one or print mixed filaments.
Thanks for taking the time to clarify this.

Ghostkeeper, 2019-07-23:
We are indeed talking about the same problem. You added an override to your profile long ago, that sets the Default Printing Temperature. That means it's no longer using the Default Printing Temperature of the material. There is no way to remove this override in the interface right now.

Similarly, if you override the Printing Temperature it will always use that override to print with. In fact this holds for all settings. Some printer's profile sets change like 50 settings if you swap materials. If you made ANY changes at all in a profile, your profile will be leading.

There are currently 3 ways to remove an override from a profile:
1. You can simply create a new profile that doesn't override the setting.
2. Clicking the (fx) button restores a setting value to its latest inheritance function. This doesn't work for settings that don't depend on other settings, like Default Printing Temperature, but it causes Printing Temperature to restore fine, for instance.
3. Edit the profile in the .cfg files.

If there is a simple way in which this can be made available through the interface, we would've made it. But we've hidden the entire system of stacked profiles from the user as well because it is too complex for an interface. It is not simple to add this without confusing people. I'd argue that the (fx) button was also a mistake in that regard.

ksihota, 2019-07-23:
Okay thanks for the clarification Ghostkeeper. 

As long as I know I can edit the cfg files to deal with issues I'm fine with that. Its easier than having to rebuild a profile from scratch and its not something that has to be done often.
Besides it's sometimes easier to read the cfg and see what's set all in one spot, rather than having to jump around in the UI.
My profiles are usually set up for specific model designs that require special handling. For most models I use a standard profile anyways.

ipatch, 2020-04-20:
@ksihota 

thank you so much for sharing this. i manage an array of materials and noticed that was having manually update the temperature in my profile everytime i wanted to use a different filament which became annoying to say the least. and also led to creating a duplicate profiles one for PETG PLA etc etc.

with your mention of editing the `.cfg` files and setting up temperatures based on the filaments / materials in cura i now have one profile for `.2` nozzle that can be used for an array of filaments / materials in cura, thus making life a lot easier.

##

#ForFutureMe 👴🏼

as of cura 4.5 running on macOS cura stores its settings in the below directory

```shell
$HOME/Library/Application\ Support/cura/
```

the last dir will depend on the version of cura one would like to edit.
> for my use case i'm running cura v4.5

within the

```
$HOME/Library/Application Support/cura/4.5/ 
```

i was able to use ripgrep / grep to search for

```
material_bed_temperature
material_print_temperature
```

and change the hardcoded values in the two different files that contained those configuration settings with

```
material_print_temperature = =default_material_print_temperature
material_bed_temperature = =default_material_bed_temperature
```

I then clicked the little rewind arrow ⃔ in the profile settings, and experimented with changing the material for the loaded object, and cura would update the temperature settings for the profile with those defined in the material settings 🚀.

> for my particular use case the two above search terms / settings were located within the `quality_changes` directory.

cheers 🍻

jonathan-schmieg, 2020-09-28:
turns out deleting the line with material_print_temperature/material_bed_temperature is enough

JohnTasto, 2020-11-17:
> to be able to remove overrides goes so deep in expert territory

The reason it is in expert territory is that it requires editing config files and the documentation to do so is hard to find.

> If there is a simple way in which this can be made available through the interface, we would've made it. But we've hidden the entire system of stacked profiles from the user as well because it is [...]

I'd argue the (fx) button is _not_ complicated, and an (unset) button wouldn't be either. The only thing I find confusing about the (fx) button is that there is no way to enter a custom formula.

> Everything should be made as simple as possible, but not simpler. 
> --commonly attributed to Albert Einstein

The profile interface in Cura has been simplified too far. When settings have no effect due to a hidden profile in the hierarchy, it is confusing. When settings cannot be unset without creating a new quality change profile from scratch, it is frustrating.

Cura's users are scientists, engineers, designers, and makers who are capable of understanding complex systems and have no desire to be babied.

Ghostkeeper, 2020-11-19:
> The reason it is in expert territory is that it requires editing config files and the documentation to do so is hard to find.

The reason it is in expert territory is because most users, even advanced users, tend to develop profiles that override a few settings for their particular use case (like some more extreme reduction of infill to improve printing speed for visual-only prints) or to change trade-offs that they themselves find more annoying (like Initial Layer Bottom Pattern to concentric to improve bed adhesion). They don't need to make settings depend on other settings for those things. Yes, we've done user research on this.

For the need to make settings depend on other settings, please refer to an open issue: https://github.com/Ultimaker/Cura/issues/3856 . Not a whole lot of people will see your arguments in this closed issue.

devTechi, 2022-08-01:
> @ksihota
> 
> 
> #ForFutureMe 👴🏼
> 
> as of cura 4.5 running on macOS cura stores its settings in the below directory
> 
> ```shell
> $HOME/Library/Application\ Support/cura/
> ```
> 
> the last dir will depend on the version of cura one would like to edit.
> 
> > for my use case i'm running cura v4.5
> 
> within the
> 
> ```
> $HOME/Library/Application Support/cura/4.5/ 
> ```
> 
> i was able to use ripgrep / grep to search for
> 
> ```
> material_bed_temperature
> material_print_temperature
> ```
> 
> and change the hardcoded values in the two different files that contained those configuration settings with
> 
> ```
> material_print_temperature = =default_material_print_temperature
> material_bed_temperature = =default_material_bed_temperature
> ```

Thanks. I was also facing this design and didn't want to recreate my profiles, so I also searched your mentioned path and found it in subfolder for the _version_, for me _5.1/quality_changes/_.

There I changed two files:
* _*.inst.cfg_
*  _*+%232.inst.cfg_

But I didn't add the variable. I just _removed_ the whole line and now the material temperature is used.

jeromeajot, 2022-11-19:
> Thanks. I was also facing this design and didn't want to recreate my profiles, so I also searched your mentioned path and found it in subfolder for the _version_, for me _5.1/quality_changes/_. [...]

I faced the same problem with v5.2.1

I made the following updates:

*_extruder_*.inst.cfg
material_print_temperature = =default_material_print_temperature
material_print_temperature_layer_0 = =default_material_print_temperature + 5

*.inst.cfg
material_bed_temperature = =default_material_bed_temperature
material_bed_temperature_layer_0 = =default_material_bed_temperature + 5

Cat-Lady, 2023-09-17:
@Ghostkeeper
The design is broken and argumentation defending the breakage is ultimately flawed.
As of current version, I can easily revert any extruder temperature overrides (the "f" icon), and it will happily revert to using material temperature, no matter how deep in the profile settings inheritance I am.

Same can't be done for bed temperature, "because". It is not only extremely frustrating, but also completely illogical. The extent of custom changes made to profiles by "most users" have nothing to do with it, it is literally unrelated to two things of same kind acting completely differently and user-unfriendly.

hapklaar, 2024-06-14:
Couldn't have said it better. It's just weird to have to go through manually editing quality config files to revert to using material values for specifically bed temperature. Shouldn't be that hard to just show a button that does that for the user?

ArixZajicek, 2024-06-27:
Really feel like this should be re-opened. The fact that extruder temperature may be unset through the UI but bed temperature cannot be is just bad UX. In this case, they should function exactly the same. Either don't let us unset the extruder temperature through the UI, or add the "Restore to Calculated Value" button to the bed temperature. That is exactly what the rest of the UI implies should exist. 
[image]

@Ghostkeeper Can I place a bounty on this? I feel pretty strongly that this is wrong and does not fit with #3856. That's a new feature request. This is not.