Ultimaker/Cura #2654: Feature request: Initial Layer temperatures based on Material temperatures plus x value

Muralha, 2017-10-21:
Instead of having to, every time change the Printing Temperature Initial Layer and Build Plate Temperature Initial Layer when changing Material, it would be better to change the extra degrees of the Initial Layer temperatures from the Material temperatures.   

Printing Temperature Initial Layer = Printing Temperature (from Material) + x value (saved in Profile?)
Build Plate Temperature Initial Layer = Build Plate Temperature (from Material) + x value (saved in Profile?)

That would make it simpler and more logical to add better adhesion to the initial layer. 
Example: 
**PLA**
Printing Temperature: 200
Printing Temperature Initial Layer: +10
Build Plate Temperature: 50
Build Plate Temperature Initial Layer: +10

ChrisTerBeke, 2017-10-21:
We're actually working on something to add an 'offset' value to any parameter which can be stored in a quality or variant profile (with a default of 0). We'll post some info in this issue thread as we have more information.

Ghostkeeper, 2018-10-30:
No, it wasn't implemented in the end because it made our settings system more complex than it already is. We're still looking for ways to reduce the problem of dimensional explosion of profiles with every configurable part of your printer being added, and this was a candidate. But it was not accepted as a solution.

Ghostkeeper, 2018-10-30:
You can turn your Initial Layer Build Plate Temperature into a formula by editing the actual profile with a text editor. To do that, open Cura, press Help -> Show Configuration Folder and close Cura. Then under quality_changes, find the profile that you'd like to edit and set the `material_bed_temperature_layer_0` setting to this:

    material_bed_temperature_layer_0 = =material_bed_temperature + 10

(Notice the double `=`.)

Adding an interface element to do this sort of thing is very tricky so don't expect that to be implemented soon.

nirurin, 2020-06-01:
Just to reopen this, as I believe it's still not been added, however - 

I think the workaround of adding the "=material_bed....etc" code to the file is fine, however... is it actually still working in the current version? 

I ask because I have my profile set to have the initial layer being +5c above the normal material temp. But when I change materials, the 'initial layer' value doesn't seem to change... so I'm not sure when/how it's 'calculating' the value, but it's not relative to the actual material temp in the box (it would seem). 

Is there a fix for this?

Ghostkeeper, 2020-06-03:
> I think the workaround of adding the "=material_bed....etc" code to the file is fine, however... is it actually still working in the current version?

Yeah, it still works. For instance, the Polypropylene profile for the Ultimaker 3 uses your exact case:

https://github.com/Ultimaker/Cura/blob/6d1d6ca1aa74a54492b964b6ec175fa0900e4613/resources/quality/ultimaker3/um3_aa0.4_PP_Normal_Quality.inst.cfg#L32

Keep in mind that if your printer has multiple extruders, the highest of them will be used as the bed temperature. So if you have one extruder that says that the initial layer bed temperature should be 60+5, but the other says it should be 9999, the initial layer bed temperature will still be 9999 no matter what your profile says.

GregValiant, 2024-11-14:
Can this request be closed?  The current versions of Cura (from 5.8.0) have logic capability in the StartUp Gcode and the Marketplace plugin "Material Settings" allows you to add any material setting to the material.  When you change material the Initial Layer Temperature (bed or nozzle) will update.

closed 2024-12-10 by github-actions[bot]