Ultimaker/Cura #2655: Cura 3.0.3 Crashes upon startup/select printer

devils27, 2017-10-22:
Hey guys, the ultimaker.com forum is no use, no one cares over there so figure I'd try with the real devs.  I downloaded new Cura 3.0.3 and I originally had trouble opening it but now I can open it after deleting local and roaming appdata cura folders.  Now I can open it but now when I'm at the print selection screen, as soon as I click ok to select the printer the program closes/crashes without any error message popup or other details. Cura 2.7 and everything before works fine.  So far to try and fix I have: updated all of windows updates, uninstalled and reinstalled all C++ ever made for x86 and 64bit, uninstalled all cura's (old and new) deleted all program and appdata folders and did a ccleaner of windows and also registry to get rid of any trace of cura leftover, and uninstalled and reinstalled all integrated Intel HD graphics drivers then re-downloaded and reinstalled cura 3.0.3 with same issue.  Windows event viewer shows nothing relevant.  Please, please please, any help would be greatly appreciated and ill try anything.  I was waiting for new Cura since last release and now I cant use it.    Thanks in advance

See logs here: seems to have errors with python scripts, especially backend.py

Cura log from "local" https://www.scribd.com/document/362122251/Cura-log-txt

stdder log from "roaming" https://www.scribd.com/document/362122194/stderr-log-txt

fieldOfView, 2017-10-22:
Could you try replacing C:\Program Files\Cura 3.0\resources\shaders\grid.shader with this file:
https://raw.githubusercontent.com/Ultimaker/Cura/1313e01f04838ac0a2d83b332dc57b6f5db0b87c/resources/shaders/grid.shader

lahines100, 2017-10-23:
I have the same problem. I replaced the grid.shader as suggested above. The program loads and displays the graphical interface but quits as soon as any change is made. I have enclosed the last few lines of the cura.log.
017-10-23 14:51:58,892 - INFO - FirmwareUpdateChecker.FirmwareUpdateChecker._onContainerAdded [38]: You have a 'FABtotum Personal Fabricator' in printer list. Let's check the firmware!
2017-10-23 14:51:58,970 - DEBUG - UM.Settings.ContainerRegistry.load [228]: Loading data into container registry took 22.819305181503296 seconds
2017-10-23 14:51:59,550 - DEBUG - UM.Controller.setActiveView [84]: Setting active view to SolidView
2017-10-23 14:51:59,992 - DEBUG - UM.Mesh.MeshData.calculateNormalsFromVertices [425]: Calculating normals took 0.0 seconds
2017-10-23 14:52:00,054 - DEBUG - STLReader.STLReader.read [79]: Loaded a mesh with 21378 vertices
2017-10-23 14:52:00,188 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [392]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.08600497245788574 seconds. 78 input vertices. 78 output vertices.
2017-10-23 14:52:13,099 - DEBUG - UM.Qt.Bindings.Theme.load [163]: Loading theme file: C:\Program Files\Cura 3.0\resources\themes\cura-light\theme.json
2017-10-23 14:52:13,900 - DEBUG - UM.Qt.Bindings.Theme.load [231]: Loaded theme C:\Program Files\Cura 3.0\resources\themes\cura-light
2017-10-23 14:52:16,650 - DEBUG - UM.Controller.setActiveView [84]: Setting active view to SolidView
2017-10-23 14:52:20,357 - WARNING - UM.Qt.Bindings.Theme.getIcon [109]: No icon tool_icon.svg defined in Theme
2017-10-23 14:52:21,826 - DEBUG - OctoPrintPlugin.DiscoverOctoPrintAction._createAdditionalComponentsView [218]: Creating additional ui components for OctoPrint-connected printers.
2017-10-23 14:52:22,031 - DEBUG - UM3NetworkPrinting.DiscoverUM3Action._createAdditionalComponentsView [138]: Creating additional ui components for UM3.
2017-10-23 14:52:22,459 - DEBUG - UM3NetworkPrinting.DiscoverUM3Action.startDiscovery [38]: Starting printer discovery.
2017-10-23 14:52:24,344 - DEBUG - UM.Qt.QtRenderer._initialize [172]: Support for Vertex Array Objects: True
2017-10-23 14:52:24,441 - DEBUG - UM.View.GL.OpenGL.__init__ [85]: Initialized OpenGL subsystems.
2017-10-23 14:52:24,497 - DEBUG - UM.View.GL.OpenGL.__init__ [86]: OpenGL Version:  4.1.0 NVIDIA 385.69
2017-10-23 14:52:24,551 - DEBUG - UM.View.GL.OpenGL.__init__ [87]: OpenGL Vendor:   NVIDIA Corporation
2017-10-23 14:52:24,611 - DEBUG - UM.View.GL.OpenGL.__init__ [88]: OpenGL Renderer: GeForce GTX 960/PCIe/SSE2
2017-10-23 14:52:24,695 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 3.0\resources\shaders\default.shader]...
2017-10-23 14:52:24,772 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 3.0\resources\shaders\selection.shader]...
2017-10-23 14:52:24,848 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 3.0\resources\shaders\default.shader]...
2017-10-23 14:52:24,925 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 3.0\resources\shaders\composite.shader]...
2017-10-23 14:52:25,007 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 3.0\resources\shaders\overhang.shader]...
2017-10-23 14:52:25,093 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 3.0\resources\shaders\striped.shader]...
2017-10-23 14:52:25,174 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 3.0\resources\shaders\default.shader]...
2017-10-23 14:52:25,254 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 3.0\resources\shaders\grid.shader]...
2017-10-23 14:52:25,354 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 3.0\resources\shaders\platform.shader]...
2017-10-23 14:52:35,753 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-10-23 15:06:29,802 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-10-23 15:07:11,723 - DEBUG - UM.Mesh.MeshData.calculateNormalsFromVertices [425]: Calculating normals took 0.0 seconds
2017-10-23 15:07:11,803 - DEBUG - STLReader.STLReader.read [79]: Loaded a mesh with 66 vertices
2017-10-23 15:07:11,840 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [392]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.06400346755981445 seconds. 8 input vertices. 8 output vertices.
2017-10-23 15:07:11,905 - DEBUG - UM.FileHandler.ReadFileJob.run [68]: Loading file took 0.8 seconds
2017-10-23 15:07:11,998 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [392]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0500030517578125 seconds. 8 input vertices. 8 output vertices.
2017-10-23 15:07:12,198 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [232]: Attempting to kill the engine process
2017-10-23 15:07:12,225 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [238]: Killing engine process
2017-10-23 15:07:12,270 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [241]: Engine process is killed. Received return code 1
2017-10-23 15:07:12,347 - DEBUG - UM.Backend.Backend._createSocket [214]: Previous socket existed. Closing that first.
2017-10-23 15:07:12,534 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [C:\Program Files\Cura 3.0\resources\shaders\transparent_object.shader]...
2017-10-23 15:07:12,750 - DEBUG - UM.Backend.Backend._logSocketState [174]: Socket state changed to Listening

devils27, 2017-10-23:
OHHH MAN I LOVE YOU fieldofview.  It worked!!!  i must have spent 10+ hrs trying to get this to work and you solved it with 1 file.  What is the problem with my shader file that yours solved?  Whats the difference?

fieldOfView, 2017-10-23:
The grid.shader file contains shaders for two different versions of OpenGL. The version of the shader that ships with Cura 3.0.3 had the correct shader for OpenGL 4.1 or newer, but a broken shader for GPUs that don't support that version of OpenGL (falling back to OpenGL 2).

Don't thank me too hard, because I wrote the broken shader in the first place. I did not notice it was broken because my computer supports the newer OpenGL shader. @daid fixed the legacy shader here:
https://github.com/Ultimaker/Cura/pull/2561
You may note that the fix was made before Cura 3.0.3 was released. Unfortunately the fix was not included in the 3.0.3 release due to an oversight.

devils27, 2017-10-23:
awesomee  thanks!. looking through the logs i knew it had something to do with OpenGL.   though one of my submitted fixes in the past aren't included in the new cura like they said.  They were suppose to allow negative retraction restart length values, sadly its still not fixed but thanks for getting it working for me.  I posted it in the ultimaker forum since there was a ton of people there too but nothing was solved, until now

daid, 2017-10-24:
Note, for me it didn't crash Cura. Else I would have pushed petty hard after learning that it was not included in 3.0.3

davebelleville, 2017-10-24:
Cura now loads and I can run through the menus but the graphics are not displayed. I'm flying blind.

davebelleville, 2017-10-24:
**Thanks! It works!**

Ghostkeeper, 2017-10-30:
The current plan, afaik, is to make a 3.0.4 build that has the fixed shader file.

Ghostkeeper, 2017-11-05:
Did 3.0.4 fix it for you?

Unconquered1, 2017-11-19:
Hi, thanks for all the hard work you put into this software. I really appreciate it. I was having the same crash issue on launch. I updated the grid.shader file as directed by fieldOfView worked great for one launch, now I get a crash on launch again. I double checked the grid.shader file again and it is still the on fieldOfView linked as the fix. This is my first time using Cura so I hope I included the correct log files for you.  Thanks!
[MyCuraInfo.zip](https://github.com/Ultimaker/Cura/files/1485314/MyCuraInfo.zip)

Ghostkeeper, 2017-11-26:
The Cura log says nothing, indicating it is a fault that's bad enough that Cura doesn't have time to write anything to the log.

The system info report indicates this line:

    11/19/2017 5:10 AM	Application Error	Faulting application name: Cura.exe, version: 0.0.0.0, time stamp: 0x59ad373a&#x000d;&#x000a;Faulting module name: ig7icd64.dll, version: 9.17.10.4229, time stamp: 0x55653f7e&#x000d;&#x000a;Exception code: 0xc0000005&#x000d;&#x000a;Fault offset: 0x0000000000740f4e&#x000d;&#x000a;Faulting process id: 0x224c&#x000d;&#x000a;Faulting application start time: 0x01d360f4a5eb4275&#x000d;&#x000a;Faulting application path: C:\Program Files\Ultimaker Cura 3.0\Cura.exe&#x000d;&#x000a;Faulting module path: C:\WINDOWS\SYSTEM32\ig7icd64.dll&#x000d;&#x000a;Report Id: 76fdc7a6-25b3-4d16-a35d-4583dd510d4d&#x000d;&#x000a;Faulting package full name: &#x000d;&#x000a;Faulting package-relative application ID:	

There's two relevant pieces of information there:
* `Exception code: 0xc0000005`, which is a permission error (Something is accessing some piece of memory that it isn't allowed to).
* `Faulting module path: C:\WINDOWS\SYSTEM32\ig7icd64.dll`. This is the Intel graphics driver.

Looking online at issues with ig7icd64.dll, it seems like it might help to update graphics drivers. They have had such crashes from time to time, fixed them, then they came back again, they fixed them again... It doesn't seem to be anything that Cura can do anything about, especially since it's unknown what OpenGL features cause this error.

Unconquered1, 2017-11-27:
Hi Ghostkeeper, thanks for the heads- up on that. I re-installed windows and forgot what a pain this early gen intel graphics chipset was. Manually updating the driver to the latest Intel driver (not the HP approved one that installs automatically) fixed the issue.

Ghostkeeper, 2017-12-11:
I'll assume then that 3.0.4 fixed the issue for the original author of this issue as well.

closed 2017-12-11 by Ghostkeeper