Ultimaker/Cura #2656: Sacrificial Bridging for overhanging holes not printed correctly

PvtPrivate, 2017-10-22:
There is an old trick when designing parts for FDM printing, one of them is creating a sacrificial bridge to avoid supports on overhanging holes. More details at https://hackaday.com/2017/10/17/sacrificial-bridge-avoids-3d-printed-supports/

Other slicers detects the bridge and prints it correctly, but cura further separates the bridge. Printing the skin on nothing.

[image]
[image]
[image]
[image]

This makes overhanging holes for countersunk screws near impossible to print.

smartavionics, 2017-10-22:
I use that technique with no problems and I can't really see what the problem is from your above images but have you tried the expand skins into infill option? If you set that to a few mm the skin that goes under your countersunk hole should be expanded out into the infill region and be well anchored.

smartavionics, 2017-10-22:
Also, a recent addition setting added to Cura is a thing called "extra skin wall count" which, by default, has been set to 1. This puts extra walls around the perimeter of a skin region which in your example above reduces the contact area of the skin itself so it's only supported by the width of 1 wall. The idea behind extra skin wall count is that when the skin is being printed on sparse infill like you have, surrounding it first with a wall can stop the skin drooping when it finishes within the infill gaps. It's a new option and I don't use it (yet) but that's their rational behind it.

PvtPrivate, 2017-10-22:
On the second picture and the fourth picture, you can see there are some travel moves forming another circle. That inner circle is printing in mid air. On the fourth picture you can clearly see the smaller circle with a different color.

S3D:
[image]
[image]

Cura:
[image]
[image]

PvtPrivate, 2017-10-22:
Found out that this problem only occurs on the last top skin of the bridge, so if i set the sacrificial bridge to be 2 layers thick, the first layer is fine.

[image]
[image]
[image]

I used to design my sacrificial bridge to be only one layer thick, which is why i noticed this problem.

smartavionics, 2017-10-22:
There is another setting called Top Surface Skin Layers. When that is non-zero, it has the effect that you have observed. I have a model that uses this technique and when Top Surface Skin Layers is zero, it prints just fine, setting it to 1 causes the same problem as you have seen:

[image]

PvtPrivate, 2017-10-23:
You are right. The problem disappear when i turned off Top Surface Skin Layers.

filipgoc, 2017-10-28:
Nice work figuring this out guys!

closed 2017-11-29 by ChrisTerBeke