Ultimaker/Cura #2657: Cura 3.03

Ruudfrezza18, 2017-10-22:
Hi,
Cura 3.03 is unable to load files f big size. it says it is loaded on Platform but nothing shows up and it can't be sliced.

smartavionics, 2017-10-22:
If the model is too big for the build plate it will be positioned off to one side - zoom out and you should see it and then you can select it and scale it to fit.

Ruudfrezza18, 2017-10-22:
I was talking about the file size in MB. When a file is too heavy it won't load. I already know how to deal with file too big for the platform. Thanks anyway

smartavionics, 2017-10-22:
Ah, OK, sorry can't help you there - what size is the file (mb)?

Ruudfrezza18, 2017-10-22:
one is 51 MB the ther 96 MB

Vandrasc, 2017-10-23:
That is weird indeed, I tried loading a 222 MB file and it successfully loaded on the platform. Could you please upload your files here so I can give it another try? Thank you!

Ghostkeeper, 2017-10-31:
Best not to upload it here on Github then but share on a third-party website such as WeTransfer.

If the model should not be shared on the public internet we'd have to find some other way to reproduce it, because it doesn't happen for me either.

closed 2018-03-02 by ChrisTerBeke