Ultimaker/Cura #2659: Print not starting on build plate

BagelOrb, 2017-10-23:
Due to a recent change (https://github.com/Ultimaker/CuraEngine/commit/51df24713500d2370e2afa7d72f0a87b9858e595) models can start to print at the second layer - even when drop on buildplate is on.

Depending on the geometry of the model the first layer might be too small to be printed. In such cases there will be no first layer, no platform adhesion and basically a failed print.

Removing empty layers in the engine should be a user setting or it should always be enabled.

The use case for not having it enabled is very small compared to the amount of failed prints caused by not removing empty layers.

ChrisTerBeke, 2017-10-23:
We can make it a setting with default to True? Right now there is no way to customise this behaviour, and it was recently changed to accommodate another GitHub issue. Adding a slice setting "Remove empty layer from bottom" should keep both parties happy.

BagelOrb, 2017-10-23:
I agree.

There has been some discussion in the past about whether that setting should always be the same as the Drop on Build Plate preference in Cura frontend....
I don't really care how this behaves, as long as there is a user setting for it.

ChrisTerBeke, 2017-10-23:
Devs: CURA-4481

ChrisTerBeke, 2017-10-31:
Now implemented in master branch.

closed 2017-10-31 by ChrisTerBeke

alekseisasin, 2017-11-01:
About "Remove empty layer from bottom", Can we change the tooltip message.
Instead of: "Remove empty layers beneath the first printed layer if they are present."
Something like this: This setting allows print models above the build plate, skipping layers if they are present.

BagelOrb, 2017-11-01:
Don't ask me. This is not a PR....