Ultimaker/Cura #2662: Cost of print behind print button

AnaGallego86, 2017-10-23:
Hello!

I just updated to Cura 3.0.

I decided to assign price for the different filaments to Cura gives me the price of the print.

If I decide to use PVA as support the price total estimation becomes to large, so that the liner appears right behind the PRINT OVER NETWORK button. I can read the total cost but you kind of have to guess a little bit.

[image]

Is it just me?

ChrisTerBeke, 2017-10-26:
Nope, I've seen that as well, but not really using the print cost feature though. Overall this part of the UI needs some re-thinking (e.g. why is the label "Print time" below the actual print time?

ChrisTerBeke, 2017-10-31:
Fixed in above referenced pull requests, due for next release.

closed 2017-10-31 by ChrisTerBeke