Ultimaker/Cura #2663: remove profile freezes interface

ChrisRiddell, 2017-10-23:
When removing a profile from preferences the interface becomes unresponsive for up to 20 seconds. 

Cura 3.0.3
Windows 10 (1709)

ChrisTerBeke, 2017-10-26:
For me removing a profile is actually quite fast (compared to removing a printer for instance). Can you put down the steps you were taking so we can accurately reproduce your workflow to see if we have an issue?

ChrisRiddell, 2017-10-26:
sorry I don't have a screen recorder on this computer so I done it on my phone but it's easier to show you this way.

https://youtu.be/1BlDGUji_ng

basically I go to manage > click the profile > remove > program becomes unresponsive

ChrisTerBeke, 2017-10-26:
Mm, haven't seen that before. Could you share the Cura logs with us so we can see if there's an error occurring?

ChrisRiddell, 2017-10-26:
[cura.log](https://github.com/Ultimaker/Cura/files/1417939/cura.log)

ChrisTerBeke, 2017-10-26:
Not seeing any errors in there. Can you share the logs from after you've tried to delete a profile?

ChrisRiddell, 2017-10-26:
I'm not sure I understand I opened cura made a profile and deleted it then closed cura and that's that log from doing that.

ChrisRiddell, 2017-10-26:
Deleted the profile and checked the log as soon as it was deleted and this is what was logged 

`
2017-10-26 21:07:46,840 - DEBUG - cura.Settings.ContainerManager.isContainerUsed [335]: Checking if container ultimaker2_plus_normal #2 is currently used
2017-10-26 21:07:49,329 - DEBUG - cura.Settings.ContainerManager.removeQualityChanges [593]: Attempting to remove the quality change containers with name Normal #2
2017-10-26 21:08:01,588 - DEBUG - UM.Settings.ContainerRegistry.removeContainer [262]: Removed container ultimaker2_plus_normal #2
2017-10-26 21:08:01,592 - WARNING - cura.Settings.QualitySettingsModel._update [100]: Could not find a quality container with id ultimaker2_plus_normal #2
2017-10-26 21:08:01,594 - DEBUG - cura.Settings.ContainerManager.isContainerUsed [335]: Checking if container ultimaker2_plus_normal #2 is currently used
2017-10-26 21:08:01,617 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-10-26 21:08:01,834 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [232]: Attempting to kill the engine process
2017-10-26 21:08:01,834 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [238]: Killing engine process
2017-10-26 21:08:01,837 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [241]: Engine process is killed. Received return code 1
2017-10-26 21:08:01,839 - DEBUG - UM.Backend.Backend._createSocket [214]: Previous socket existed. Closing that first.
2017-10-26 21:08:02,047 - DEBUG - UM.Backend.Backend._logSocketState [174]: Socket state changed to Listening`

ChrisTerBeke, 2017-10-26:
There's indeed around 10 seconds of nothing between attempting and actually removing the quality profile.

```
2017-10-26 21:07:49,329 - DEBUG - cura.Settings.ContainerManager.removeQualityChanges [593]: Attempting to remove the quality change containers with name Normal #2
2017-10-26 21:08:01,588 - DEBUG - UM.Settings.ContainerRegistry.removeContainer [262]: Removed container ultimaker2_plus_normal #2
```

Not sure what's causing that but we'll dive a bit into that function and see if we can find something.

ChrisTerBeke, 2017-10-26:
We looked into it and it will be fixed in the 3.1 release (due in December).

closed 2017-10-26 by ChrisTerBeke