Ultimaker/Cura #2664: 3.0.3 tests are failing

hroncok, 2017-10-23:
I'm updating the Fedora package to 3.0.3, however the tests are failing (once again, after thay have been failing in 2.6 and fixed in 2.7). Does this indicate something wrong with the packaging, or are the tests failing for you as well?

```
+ /usr/bin/python3 -m pip freeze
numpy==1.13.3
py==1.4.34
pytest==3.2.3
scipy==0.19.1
+ /usr/bin/python3 -m pytest -v
============================= test session starts ==============================
platform linux -- Python 3.6.3, pytest-3.2.3, py-1.4.34, pluggy-0.4.0 -- /usr/bin/python3
cachedir: .cache
rootdir: /builddir/build/BUILD/Cura-3.0.3, inifile: pytest.ini
collecting ... collected 170 items

tests/TestArrange.py::test_smoke_arrange PASSED
tests/TestArrange.py::test_smoke_ShapeArray PASSED
tests/TestArrange.py::test_centerFirst PASSED
tests/TestArrange.py::test_backFirst PASSED
tests/TestArrange.py::test_smoke_bestSpot PASSED
tests/TestArrange.py::test_smoke_place PASSED
tests/TestArrange.py::test_checkShape PASSED
tests/TestArrange.py::test_checkShape_place PASSED
tests/TestArrange.py::test_smoke_place_objects PASSED
tests/TestArrange.py::test_arrayFromPolygon PASSED
tests/TestArrange.py::test_arrayFromPolygon2 PASSED
tests/TestArrange.py::test_check PASSED
tests/TestArrange.py::test_check2 PASSED
tests/TestMachineAction.py::test_addMachineAction PASSED
tests/TestProfileRequirements.py::test_ultimaker3extended_variants[ultimaker3_aa0.8.inst.cfg-ultimaker3_extended_aa0.8.inst.cfg] PASSED
tests/TestProfileRequirements.py::test_ultimaker3extended_variants[ultimaker3_aa04.inst.cfg-ultimaker3_extended_aa04.inst.cfg] PASSED
tests/TestProfileRequirements.py::test_ultimaker3extended_variants[ultimaker3_bb0.8.inst.cfg-ultimaker3_extended_bb0.8.inst.cfg] PASSED
tests/TestProfileRequirements.py::test_ultimaker3extended_variants[ultimaker3_bb04.inst.cfg-ultimaker3_extended_bb04.inst.cfg] PASSED
tests/Settings/TestCuraContainerRegistry.py::test_addContainerExtruderStack PASSED
tests/Settings/TestCuraContainerRegistry.py::test_addContainerGlobalStack PASSED
tests/Settings/TestCuraContainerRegistry.py::test_addContainerGoodSettingVersion PASSED
tests/Settings/TestCuraContainerRegistry.py::test_addContainerNoSettingVersion PASSED
tests/Settings/TestCuraContainerRegistry.py::test_addContainerBadSettingVersion PASSED
tests/Settings/TestCuraContainerRegistry.py::test_loadTypes[ExtruderLegacy.stack.cfg-ExtruderStack] PASSED
tests/Settings/TestCuraContainerRegistry.py::test_loadTypes[MachineLegacy.stack.cfg-GlobalStack] PASSED
tests/Settings/TestCuraContainerRegistry.py::test_loadTypes[Left.extruder.cfg-ExtruderStack] PASSED
tests/Settings/TestCuraContainerRegistry.py::test_loadTypes[Global.global.cfg-GlobalStack] PASSED
tests/Settings/TestCuraContainerRegistry.py::test_loadTypes[Global.stack.cfg-GlobalStack] PASSED
tests/Settings/TestCuraContainerRegistry.py::test_loadLegacyFileRenamed PASSED
tests/Settings/TestExtruderStack.py::test_addContainer PASSED
tests/Settings/TestExtruderStack.py::test_constrainUserChangesInvalid[container0] PASSED
tests/Settings/TestExtruderStack.py::test_constrainUserChangesInvalid[container1] PASSED
tests/Settings/TestExtruderStack.py::test_constrainUserChangesInvalid[container2] PASSED
tests/Settings/TestExtruderStack.py::test_constrainUserChangesValid[container0] PASSED
tests/Settings/TestExtruderStack.py::test_constrainUserChangesValid[container1] PASSED
tests/Settings/TestExtruderStack.py::test_constrainQualityChangesInvalid[container0] PASSED
tests/Settings/TestExtruderStack.py::test_constrainQualityChangesInvalid[container1] PASSED
tests/Settings/TestExtruderStack.py::test_constrainQualityChangesInvalid[container2] PASSED
tests/Settings/TestExtruderStack.py::test_constrainQualityChangesValid[container0] PASSED
tests/Settings/TestExtruderStack.py::test_constrainQualityChangesValid[container1] PASSED
tests/Settings/TestExtruderStack.py::test_constrainQualityInvalid[container0] PASSED
tests/Settings/TestExtruderStack.py::test_constrainQualityInvalid[container1] PASSED
tests/Settings/TestExtruderStack.py::test_constrainQualityInvalid[container2] PASSED
tests/Settings/TestExtruderStack.py::test_constrainQualityValid[container0] PASSED
tests/Settings/TestExtruderStack.py::test_constrainQualityValid[container1] PASSED
tests/Settings/TestExtruderStack.py::test_constrainMaterialInvalid[container0] PASSED
tests/Settings/TestExtruderStack.py::test_constrainMaterialInvalid[container1] PASSED
tests/Settings/TestExtruderStack.py::test_constrainMaterialInvalid[container2] PASSED
tests/Settings/TestExtruderStack.py::test_constrainMaterialValid[container0] PASSED
tests/Settings/TestExtruderStack.py::test_constrainMaterialValid[container1] PASSED
tests/Settings/TestExtruderStack.py::test_constrainVariantValid[container0] PASSED
tests/Settings/TestExtruderStack.py::test_constrainVariantValid[container1] PASSED
tests/Settings/TestExtruderStack.py::test_constrainVariantInvalid[container0] PASSED
tests/Settings/TestExtruderStack.py::test_constrainVariantInvalid[container1] PASSED
tests/Settings/TestExtruderStack.py::test_constrainDefinitionValid[container0] PASSED
[... 415 lines cut ...]
key = 'material_bed_temperature', property_name = 'state'

    def getRawProperty(self, key, property_name, *, context: Optional[PropertyEvaluationContext] = None,
                       use_next = True, skip_until_container = None):
        containers = self._containers
        if context is not None:
            # if context is provided, check if there is any container that needs to be skipped.
            start_index = context.context.get("evaluate_from_container_index", 0)
            if start_index >= len(self._containers):
                return None
            containers = self._containers[start_index:]
    
        for container in containers:
            if skip_until_container and container.getId() != skip_until_container:
                continue #Skip.
            skip_until_container = None #When we find the container, stop skipping.
    
>           value = container.getProperty(key, property_name, context)
E           TypeError: <lambda>() takes 2 positional arguments but 3 were given

/usr/lib/python3.6/site-packages/UM/Settings/ContainerStack.py:240: TypeError
=============== 7 failed, 159 passed, 4 skipped in 51.49 seconds ===============
```

That's python 3.6.3 on Fedora 28 x86_64.

ChrisTerBeke, 2017-10-26:
Our nightly automated builds run fine and include running the tests. We make builds for all OSs, but not sure what the Linux distro is (probably Debian based).

awhiemstra, 2017-10-26:
No, they do not. The nightly builds just build. See http://hal9000.ultimaker.com:29130/job/Ultimaker/job/Cura/job/master/360/testReport/ for the test results, which confirms @hroncok's observation.

ChrisTerBeke, 2017-10-26:
OK, we'll need to fix some of the tests then. Devs: CURA-4498

LipuFei, 2017-10-26:
Made a fix for this: 7d46eead606170b7fab9c1f1e51b10caf554adf1

closed 2017-10-27 by hroncok