Ultimaker/Cura #2665: Fill gaps between walls is less effective in Cura 3.0

lokster, 2017-10-23:
Comparison between a model, printed with the same settings profile with Cura 2.7 and Cura 3.0.
The thin gaps between the walls of the letters are not filled as good as in 2.7
[image]

Top - Cura 3, bottom - Cura 2.7
[image]

ChrisTerBeke, 2017-10-23:
This seems the result of the new "Skin extra wall count" feature? Try setting that value to 0 and it should be the same as 2.7.

smartavionics, 2017-10-23:
Making the default value for Skin Extra Wall Count 1 and not zero has caused several issues/queries. IMHO it was a mistake to do that.

lokster, 2017-10-23:
Setting the "Extra Skin Wall Count" to 0 makes a bit of difference, but does not make the result better, as there are still more gaps than on version 2.7:
[image]

ChrisTerBeke, 2017-10-23:
@smartavionics we'll see if we can change that for 3.1. @lokster I'm not aware of any other changes in the engine that could have caused this, but I'll ask if others can take a look at this issue.

lokster, 2017-10-24:
Just a clarification - I wanted to make sure the problem is not in my settings profile, so I tested this with the default "Draft Quality" profile for "Custom FDM Printer" machine.
Setting Extra Skin Wall Count to 0 in Cura 3.0 makes a little difference - just more top skin, and still a lot of gaps. As if the fill gaps between walls is almost not working.
In Cura 2.7 the gaps are filled better, no matter if Extra Skin Wall Count is 0 or 1

smartavionics, 2017-10-24:
There is a "Fill Gaps Between Walls" setting that can be either Nowhere or Everywhere (why isn't it just a checkbox?). Perhaps that is set to nowhere?

lokster, 2017-10-24:
No, it's set to "Everywhere" on both versions.
Also, I tried setting the "Ignore Small Z gaps" to 0 (It defaults to 1 on the default profile, but I usually set it to 0) - it made no difference.

smartavionics, 2017-10-24:
Can you please post a link to the model so I can have a play?

lokster, 2017-10-24:
Yes, I've just put it on thingiverse - https://www.thingiverse.com/thing:2603330

lokster, 2017-10-24:
Sorry, accidentally closed the issue :)

smartavionics, 2017-10-24:
I agree, it definitely seems to leave gaps unfilled that in your 2.7 image show as filled. One thing that will help a little is that if you have the compensate wall overlaps enabled, disabling it will cause extra extrusion in the narrow regions which may help to fill the gaps.

It seems to me that the current cura has introduced some lower limit on the size of lines used to fill the gaps between the walls or something else that has a similar effect.

smartavionics, 2017-10-24:
Another suggestion is that you use a wall line count of 1.

smartavionics, 2017-10-24:
[image]

smartavionics, 2017-10-24:
Found where this was changed, it was 9b999e7e0.

Thisismydigitalself, 2017-10-28:
Is it only me, or others feel as i do when it comes to how simple it is to slice a model in a bad way when in fact Cura walls / infill algorithm should know better what to do?

I feel that way. 
I hope this will be taken as a constructive criticism.

smartavionics, 2017-10-28:
> Is it only me, or others feel as i do when it comes to how simple it is to slice a model in a bad way when in fact Cura walls / infill algorithm should know better what to do?
I feel that way.
I hope this will be taken as a constructive criticism.

It's open source, feel free to implement better algorithms and fix bugs and submit them for inclusion so that everyone will benefit from your efforts.
I hope this will be taken as a constructive comment.

Thisismydigitalself, 2017-10-28:
> It's open source, feel free to implement better algorithms and fix bugs and submit them for inclusion so that everyone will benefit from your efforts.
I hope this will be taken as a constructive comment.

> 

I thank you both for correcting me and pointing me in the right direction.You did the right thing.
my only regret is for the words you have chosen to express yourself and to correct me but i guess it's my fault for stepping on your toes in the first place without any intention whatsoever to do so but why should you two care? Anyway, I wont be saying what i have to say anymore. too afraid now. I will chose a place which accepts freedom of speech, opinions and ideas without fear of censorship.

smartavionics, 2017-10-28:
> Anyway, I wont be saying what i have to say anymore. too afraid now. I will chose a place which accepts freedom of speech, opinions and ideas without fear of censorship.

Censorship? I don't think so. You are free to say whatever you like and in whatever manner you wish as long as it is not offensive. Did my little bit of sarcasm upset you? Well, sorry about that but the point I was making is a real one. **Anyone can make contributions to Cura**, if you think that something should be done better then make it better. Just saying that slicing is simple and Cura should do a better job is like waving a magic wand. Unfortunately, this is the real world, not fairy land!

lokster, 2017-11-01:
This particular issue (and nothing else) is really preventing me from using the 3.0 version, so I hope it will get fixed.
None of the suggested workarounds fixes it, and they affect negatively how other parts of the model are printed. E.g. setting "wall line count" to 1 or adding a bit of over extrusion isn't really a viable solution.
It gets even worse when I use Line Width bigger than the nozzle width (e.g. bigger gaps).
Interestingly, I never had blobs on the outside walls due to the gaps being filled fully in the old versions.

smartavionics, 2017-11-01:
This really needs @BagelOrb  to justify why he did https://github.com/Ultimaker/CuraEngine/commit/9b999e7e0d39f590968a73a722725c552c3c9789. Perhaps this decision needs to be re-evaluated?

BagelOrb, 2017-11-01:
There spots are removed that are smaller than 1x2 times the line width. The idea was that a lot of time is going into filling gaps which are too small to even see or to fill properly. Gap filling is most effective when large areas need to be filled.

This is not a justification - just an explanation of my actions. Perhaps this does need re-evaluation.

smartavionics, 2017-11-01:
Thanks Tim. Well, I guess the devs need to have a discussion about this. @ChrisTerBeke ? @Ghostkeeper ?

lokster, 2017-11-01:
I do think It needs re-evaluation, because the gaps are pretty visible, especially for this kind of models which have flat letters on top or other flat features with sharp angles.

Ghostkeeper, 2017-11-05:
Yeah I agree. Last time the change was instigated purely by looking at layer view. There are no documented real-life tests with an actual print on the ticket (CURA-4230).

Let's take another look at it and maybe reduce the minimum area.

gr5, 2017-11-08:
@lokster please realize that slicers are complicated.  What seems simple to a human is sometimes quite complicated to do in real life.  The variation in models that can be sliced is astounding if you start looking carefully at designs on thingiverse and their needs are all different and often contradictory.  A tiny change to the slicer might make some models come out better and other's come out worse.  Someone else is probably very happy that Cura 3 no longer fills the tiny gaps because those movements created extra lines across the top of their part so the nozzle could go back and fill those tiny holes.

lokster, 2017-11-08:
If what you say is true, then this change must be made optional / configurable, or just linked to "fill gaps between walls". So everyone is happy.

ungutknut, 2017-11-13:
I have a similar problem, but with very different geometry. Since 3.0 narrowing corners get printed more ugly than before, because cura leaves gaps like in the pictures shown by lokster. While I understand that perhaps there was need for leaving the gaps open, I prefer the old behaviour.

jackha, 2017-11-21:
@lokster I'm looking into it and it indeed looks as if too much perimeters are removed. I'm testing and tuning a bit.
Thanks @smartavionics for finding the corresponding commit.

jackha, 2017-11-21:
OK I'm making an option out of it :-)

jackha, 2017-11-22:
I tried different thresholds from 2*sqr(line width) to 0.05*sqr(line width) to no filtering. There is no threshold that fixes @lokster 's problem and does not leave an under extrusion on some other printers. That's why I made a setting out of it: Filter out tiny gaps. The default value is "true", which is the old behaviour. This goes together with an update in Cura where the setting is added.

lokster, 2017-11-22:
Perfect! This will be really useful (at least to me, anyway), because in the mean time I've had similar issues with other models too.

lokster, 2017-12-05:
I see the setting "Filter out tiny gaps" is present in Cura 3.1. Unfortunately, it has absolutely no effect on this particular STL and a couple of others I just tried. The generated gcode is 100% identical when compared with diff, no matter if the setting is on or off.

jeraziah, 2017-12-05:
Just wanted to note that I believe this issue is more widespread than one might think. The MP select mini group has seen many models come off with significant issues that look identical to what is shown in the slicer view here. (ex: https://imgur.com/a/YdBuP , https://imgur.com/a/nigEO , and https://imgur.com/a/Buq6F)

Ghostkeeper, 2017-12-18:
The setting still works though. I've just tried with this model using a Custom FDM Printer:
[BlastWall.zip](https://github.com/Ultimaker/Cura/files/1567573/BlastWall.zip)

When Filter Out Tiny Gaps is activated, it won't fill the tiny gap between the outer and inner wall.

lokster, 2017-12-18:
Looks like I compared the wrong files, because, yeah - the gcode is different with the setting on or off.
However, even with the setting "off" the result is far worse, than the perfect result I was getting with Cura 2.7.
It's just less effective when filling gaps, and skips some of them.
Here it is - the same stl, sliced with Cura 3.1 with "Filter Out Tiny Gaps" = unchecked
[image]

As you can see, the gaps are still there, and there isn't much of a difference, compared to what I was getting with version 3 (which didn't had this setting).
It just leaves those gaps empty, no matter if I use print thin walls / compensate wall overlaps / ignore small z gaps etc. The only way to "fill" them, is to add over extrusion, which does not make sense on a perfectly tuned printer.
Can't "Filter Out Tiny Gaps" = unchecked just revert to the old behavior? Because currently, it does not.

CCS86, 2017-12-26:
I have also noticed this reduction in gap filling efficacy. Hopefully the devs can ge to the bottom of what code change caused this.

If it is a minimum gap area cutoff, perhaps it can just be changed to a user-configurable setting, instead of hard coded. Then everybody wins.

Thorinair, 2018-01-15:
Just chiming in to confirm that this issue is still a thing. Any chance of adding a toggle that actually works and reverts the functionality to pre 3.x? Would be helpful as this bug is extremely annoying.

Ghostkeeper, 2018-01-29:
Here's the print in 3.0:
[image]

Here's the print in 3.2 with filtering small gaps (default):
[image]

Here's the print in 3.2 without filtering small gaps:
[image]

So it looks like you need to keep the filter on (this is also reflected in the code) but the difference may be that there is an extra skin wall. So I suggest you set Extra Skin Wall Count to 0 and then increase the number of walls by 1.

lokster, 2018-01-29:
Perfect! I tested it (just preview, no real printing yet), and it looks OK.
I will post a picture later when I have time to print some test, but I think it will be OK. Thanks!

closed 2018-02-14 by Ghostkeeper

nicolinux, 2018-02-28:
I have an update on this issue. None of the options and their combinations yielded a good result with Cura 3.2.1. Only setting the wall count to 1 (like @smartavionics) suggested, produced a perfect print.
 
[image]
[image]