Ultimaker/Cura #2666: Variable slicing algorithm

buttigieg94, 2017-10-23:
Integrate variable slicing as explained by this video: https://www.youtube.com/watch?v=HAmneiL5-jQ

ChrisTerBeke, 2017-10-23:
There's an internal investigation into this functionality, but nothing concrete yet. We can use this issue to communicate about progress.

buttigieg94, 2017-10-23:
Would be a good idea to do so...

ChrisRiddell, 2017-10-26:
Pretty interesting video and the linked pull request for Slic3r in the video's description found it interesting in how they implemented it. 

From my understanding with some layer heights depending on type of lead screw and steppers step angle you can get errors over x amount of height in the model. 

Could this be an issue with many printers using different lead screws etc?

ChrisTerBeke, 2017-10-26:
I'm picking this topic up in the upcoming "research sprint" that starts end of next week, where I'll have 2 weeks to implement something that works and we can use to build on top of later. Any suggestions are welcome!

For now I'm thinking using the Autodesk video and the Slic3r Prusa Edition variable layer height feature as starting point.

ChrisRiddell, 2017-10-26:
I'd recommand using that video and this https://github.com/alexrj/Slic3r/pull/1386 for the starting point.

To be honest I'm not a fan of the way Slic3r Prusa Edition done it, I really like how KISSlicer implemented it where you put a min and max layer height and the engine works out the best heights over the model  based on the geometry etc, as that would feel more Cura like where it does alot for you eg how ironing was implemented.

ChrisTerBeke, 2017-10-26:
Geometry based with min and max values for layer height sounds great, as it's mostly automated. I also think it's doable to implement that in a 2 week time frame.

nickthetait, 2017-10-27:
Here's a lil mockup I made. Shows two endpoints to select min and max values.
[image]

For reference, what it looks like currently (v3.0.3)
[image]

Good luck on the research :smile:

buttigieg94, 2017-10-27:
Looks great, the only thing I recommend as an "advanced setting" is to add another window to set the layer height in relation to the layer angle i.e.
layer angle (50deg) ->Layer height (0.1)
layer angle (50deg) ->Layer height (0.1)
layer angle (50deg) ->Layer height (0.1)

Ghostkeeper, 2017-10-27:
We need to keep in mind that VariSlice itself is patented.

The greatest problem to implement this is how to get the greatest quality with the rest of the settings. The implementation of finding a layer height to slice with is easy enough, but we know that changing the layer height of the 0.15mm profile to 0.13mm tends to result in worse quality (if you have properly optimised profiles) because the rest of the settings are not optimised for a 0.13mm layer height.

I consider this a duplicate of https://github.com/Ultimaker/CuraEngine/issues/477

buttigieg94, 2017-10-27:
Was not aware of the other duplicate, however I suggest this post stays just to make it a form of reminder to implement this feature..

nickthetait, 2017-10-27:
... and more people watch the Cura project than do CuraEngine

ChrisTerBeke, 2017-10-31:
I'm keeping an eye on both threads and will go through all options when I start the project later this week.

ChrisTerBeke, 2017-11-02:
Current plan:

1) Adapt CuraEngine to allow regions input in which settings can be changed per region (including layer height). I'm sure it is complicated and brings some issues, but for now a rudimentary version is OK.
2) Port the VariSlice algorithm (which is open source under the Creative Commons license!) to Python so I can make a Cura 3 plugin that analyses the model and outputs the layer height per region (exactly what VariSlice already does but in Java/Processing).

Wish me luck! :P

ChrisTerBeke, 2017-11-02:
Plugin repo: https://github.com/ChrisTerBeke/CuraVariSlicePlugin.

nickthetait, 2017-11-02:
*is very excited*

I'm sure we can come up with a better name than "VariSlice" before stable release. :thinking: Especially since we are no longer restricted to just a single setting (layer height) nor zones restricted by Z height (horizontally). 

"regions input" seems a similar concept to the UI challenges required for manual supports. Chris' work may enable future development in that area too...

buttigieg94, 2017-11-02:
@nickthetait maybe vSlice? Sounds apple-esk minus the i-branding

ChrisRiddell, 2017-11-02:
This is great you are making this as a plugin to be honest.

mikko70, 2017-11-07:
When this happen, i hope possibility raise bed and nozzle temperature about 5 celsius for first layers. If you haven't warming bed, cold bed need raise temp first.

ChrisTerBeke, 2017-11-08:
Started WIP pull request for this feature in the engine: https://github.com/Ultimaker/CuraEngine/pull/647.

ChrisTerBeke, 2017-11-10:
First rudimentary version is now working! It only supports layer height, but coming days I'll investigate other parameters as well. The variable one uses the layer heights that are available in the quality profiles for the selected machine + material, so everything is automated!

Here's a picture of a sphere compared to one that only uses the 0.2 layer height. The print times are 31 vs 39 minutes.

[image]

nickthetait, 2017-11-10:
Woot, awesome!

ChrisRiddell, 2017-11-11:
Nice work,

How does it handle things like 3DBenchy, does the algorithm use the best layer heights (available come the quality profiles) for things like overhangs etc?

ChrisTerBeke, 2017-11-11:
@ChrisRiddell I loop over all triangles/faces in a potential layer and find the steepest one. If it's steeper than an arbitrary treshold it reduces the layer height and tries again. So depending on the wall angle it uses the appropriate layer height from the available quality profiles. The algorithm is here: https://github.com/ChrisTerBeke/CuraVariSlicePlugin/blob/master/VariSlice/VariSliceAlgorithm.py#L61. It's a bit cryptic as I rewrote everything to NumPy arrays for faster processing (e.g. comparing array indices for finding the slopes for relevant triangles etc.).

fieldOfView, 2017-11-12:
May I suggest the terminology "adaptive layer height" instead of "varislice"? "VariSlice", or even "Variable slicing" does not indicate what is variable or that there is any setting is being changed over the progression of the slice.

buttigieg94, 2017-11-12:
@fieldOfView True, and makes sense. I like that!

ChrisTerBeke, 2017-11-13:
+1 for adaptive layer heights. Right now I'm converting the Python algorithm in into c++ for the engine to have it built in and being able to attach other parameters per adaptive layer that need changing (like speed, temperature, top/bottom thickness, support z distance, etc). Next sprint I will continue to work on this functionality (so 2 weeks of extra time). So far the test prints are very promising!

ChrisTerBeke, 2017-11-14:
Printed a bigger model (14 hours) and here's the result:

[image]

Don't mind the small blobs, those are another issue. The top surfaces of the fingers are quite smooth now! Also almost done with the algorithm in c++ so the end user only has to enable "Use adaptive layer heights" in the experimental section.

fieldOfView, 2017-11-14:
Nice. There does seem to be quite a bit of banding though. I am guessing that's because there aren't all that many profiles to pick from. Interpolating between profiles could be an interesting follow-up research project.

ChrisTerBeke, 2017-11-14:
Next sprint we'll focus on getting the most important other parameters in there (temperature, print speed, top/bottom thickness) to reduce the banding. Interpolating between profiles might be a part of that. We'll ask materials for the most important parameters. Now that the calculation happens in the engine it's a lot easier to add new params to it.

ChrisTerBeke, 2017-11-14:
Also it's very obvious with this materials, the metallic particles refract the light differently per layer height.

filipgoc, 2017-11-28:
Great work @ChrisTerBeke 

I was just thinking today that the easiest (if a bit clunky) way to get variable layer height a la Simplify 3D (with sharp boundaries) would be to allow layer height as a Per-Model Setting (with a sanity check that models would have to be vertically aligned) and users could use the current multi-color model merging workflow to print different meshes with different layer height. Basically what you mention about a [month ago](https://github.com/Ultimaker/Cura/issues/2666#issuecomment-341419403). It's exciting to see this getting so much progress.

ChrisTerBeke, 2017-11-28:
Submitted the [pull request](https://github.com/Ultimaker/CuraEngine/pull/649) for CuraEngine today, it's currently in review. When it's merged to master it'll be available under the experimental settings. There's some known limitations at the moment, but for experimental it's more than good enough I believe.

ChrisTerBeke, 2018-01-29:
Closing now that it's implemented. Other threads already popped up about how to improve it in the future.

closed 2018-01-29 by ChrisTerBeke