Ultimaker/Cura #2667: Negative retraction extra prime amount

devils27, 2017-10-23:
New Cura 3.0.3 doesnt allow negative retraction restart volume to prevent blobs on restart of retraction.  There was an issue already submitted for 2.7 that I thought would be included in the new release but I see its still the same.  Can you please make sure next update includes option in code to go into negatives like what was previously submitted but never included?

thanks

smartavionics, 2017-10-24:
Hi. Does this affect Ultimaker printers? If so, it could get some attention from the Cura devs. If not, then you will need to find a 3rd party to implement and test this request and submit it as a PR. Otherwise, I don't think it will happen.

fieldOfView, 2017-10-24:
Could you add a link to where you already submitted the request for 2.7? (hoping there are more details there)

devils27, 2017-10-28:
https://github.com/Ultimaker/Cura/issues/2392

devils27, 2017-10-28:
https://github.com/Ultimaker/CuraEngine/commit/27636106f4f298332f9220513b69a975370eda53

devils27, 2017-10-28:
its as I have read for all printers, ultimaker or not.  it doesn't matter in cura which printer you use, a negative value in for extra retraction restart spot is always red and doesn't apply anything to the gcode either, its basically like putting 0 there

maukcc, 2017-10-30:
The minimum_value_warning in fdmprinter.def.json needs updating as well.

bcrens7, 2017-10-31:
Im having the same problem on mine. I have an ultimaker 3 extended and the retraction is giving me blobs on all my prints.

Ghostkeeper, 2017-11-07:
It seems to be fixed when I try it though:

[image]

It's giving a warning colour (orange), which is expected behaviour.

Ghostkeeper, 2017-11-11:
> Why is everybody getting blobs al of a sudden?

Not everybody is. It seems to be limited to some printers only.

> Should we leave the problem as is, and fix a symptom?

Probably not, but what is the problem, concretely? What modifications to the g-code would you make to prevent the problem?

maukcc, 2017-11-13:
Well, something has changed between 3.0.0 and 3.0.3 that started to make the blobs.
The prime amount looks to be good.
Has something changed with coasting?
We also see a new M82 code at the beginning, but I cant imagine that has anything to do with it.
We also see the E number increasing all through the file. It used to start with zero every layer I think?
Also "optimize wall printing order" is default off, and therefore generates much more retractions/primes on outer skin. 
We will do some testing.

ChrisTerBeke, 2017-11-13:
There was no engine or settings change between 3.0.0 and 3.0.3. We only added an SSL library for plugins and fixed a OpenGL graphics issue.

genechuah, 2017-11-13:
It goes orange when I enter a negative value, should I be concerned? Haven't experimented yet but wondering if I can get more direct info.

ChrisTerBeke, 2017-11-14:
Orange is a warning. It will print, but most likely give ugly results. You don't have to worry about damaging your machine. When it's red, it's just not allowed.

badvision, 2017-11-16:
I have the same issue.  #2795  Makes printing really gooey stuff like PETG impossible without lots of post-printing cleanup.

Ghostkeeper, 2017-11-25:
It becomes orange because you're instructing Cura to underextrude on purpose. The setting is meant to overextrude on purpose because you've just lost some material while travelling. We don't expect there to be extra material upon unretracting.

Maybe the blobs are caused by the nozzle standing still while unretracting?

badvision, 2017-11-26:
Right but the problem is that the extrusion starts too early, at least for the PETG that I'm printing, so I get strings 2mm before the starting edge following every retraction.  The retraction/wiping itself works fine, but because of the early priming, I get blobs -- and they happen while it's moving.

smartavionics, 2017-11-26:
Hi, I can't imagine how the prime extrusion is happening too early so could you please post the gcode that shows this problem (i.e. slice it without the -ve prime amount). Please provide info as to where in the print the problem is occurring. Thanks.

badvision, 2017-11-27:
For example around line 54180 (gcode attached in the dropbox folder below) it seems like it gives the command to move to the next tower at the same time it also tells it to start extruding.  I think that could be the cause of it priming too early, at least with my printer (running Repetier firmware, if that helps)

For posterity's sake I posted all relevant files to dropbox: https://www.dropbox.com/sh/wjzmtj6ldm7xns4/AADhVEc-Vq8JOle4Nhrrw2GAa?dl=0

The notes image shows the layers view, and indicates print order (clockwise) and where the blobs are appearing in the print.  I also took a picture showing the oozing on the left side -- this is not the best example because I'm also trying to adjust other things (temperature, z-hop, etc) but the oozing is consistently on the priming side of objects even when retraction seems to look good elsewhere.

smartavionics, 2017-11-27:
Hi, thanks for making those files available. Stepping through the gcode, I can't see it priming too early like you are saying. The un-retractions (primes) are occurring at the correct positions. I see that you are using very little retraction. I print PETG using a non-bowden extruder and I use 2mm of retraction with zero prime and get very little stringing. What happens if you use 2mm of retraction?

smartavionics, 2017-11-27:
I printed the model, I got some very fine stringing which I don't consider a problem, quite different from your picture:

[image]

smartavionics, 2017-11-27:
Z-seams came out OK:

[image]

badvision, 2017-11-27:
I have a microswiss all-metal hotend -- if I use more than 2mm retraction, I start to get filament grinding and run the risk of clogs.  What are your retraction and priming speeds?

smartavionics, 2017-11-27:
I'm using an e3d v6 with a flex3drive extruder. My retract/prime speeds are quite slow at 20 mm/S.

smartavionics, 2017-11-27:
BTW, I also use 0.5mm of z-hop.

smartavionics, 2017-11-27:
````
;SETTING_3 {"global_quality": "[general]\\nversion = 2\\nname = PETG XL\\ndefini
;SETTING_3 tion = custom\\n\\n[metadata]\\ntype = quality_changes\\nquality_type
;SETTING_3  = normal\\nsetting_version = 4\\n\\n[values]\\nacceleration_enabled 
;SETTING_3 = False\\nacceleration_wall = None\\nadhesion_type = skirt\\ncool_fan
;SETTING_3 _full_at_height = 4\\ncool_fan_min_at_height = 2\\ncool_fan_speed_min
;SETTING_3  = 50.0\\ncool_min_layer_time = 15\\ncool_min_layer_time_fan_speed_ma
;SETTING_3 x = 15\\ncool_min_speed = 15\\nexpand_skins_expand_distance = 2\\nfil
;SETTING_3 l_perimeter_gaps = nowhere\\ninfill_before_walls = False\\ninfill_lin
;SETTING_3 e_width = 0.8\\ninfill_overlap = 10\\ninfill_pattern = zigzag\\ninfil
;SETTING_3 l_sparse_density = 40\\njerk_enabled = False\\njerk_wall = 5\\nlayer_
;SETTING_3 height = 0.2\\nlayer_height_0 = 0.3\\nline_width = 0.5\\nmaterial_bed
;SETTING_3 _temperature = 70\\nmaterial_bed_temperature_layer_0 = 75\\nmaterial_
;SETTING_3 diameter = 1.75\\nmaterial_flow_layer_0 = 90\\nmaterial_print_tempera
;SETTING_3 ture = 240\\nmaterial_print_temperature_layer_0 = 245\\nmin_skin_widt
;SETTING_3 h_for_expansion = 0.5\\noptimize_wall_printing_order = True\\nouter_i
;SETTING_3 nset_first = False\\nretraction_amount = 2\\nretraction_combing = off
;SETTING_3 \\nretraction_extra_prime_amount = 0\\nretraction_hop = 0.5\\nretract
;SETTING_3 ion_hop_enabled = True\\nretraction_min_travel = 2.5\\nretraction_pri
;SETTING_3 me_speed = =retraction_speed\\nretraction_retract_speed = =retraction
;SETTING_3 _speed\\nretraction_speed = 20\\nskin_outline_count = 0\\nskin_overla
;SETTING_3 p = None\\nskirt_brim_minimal_length = 500\\nskirt_brim_speed = 20.0\
;SETTING_3 \nskirt_gap = 8\\nspeed_infill = 60\\nspeed_layer_0 = 40.0\\nspeed_pr
;SETTING_3 int = 50\\nspeed_slowdown_layers = 1\\nspeed_topbottom = 40.0\\nspeed
;SETTING_3 _travel = 300\\nspeed_wall_0 = =speed_wall\\nspeed_wall_x = =speed_wa
;SETTING_3 ll * 2\\nspeed_walls_layer_0 = 15.0\\nsupport_angle = 60\\nsupport_en
;SETTING_3 able = False\\nsupport_infill_rate = 25\\nsupport_interface_enable = 
;SETTING_3 True\\nsupport_interface_height = 0.6\\nsupport_pattern = zigzag\\nsu
;SETTING_3 pport_xy_distance = 1\\nsupport_z_distance = 0.3\\ntop_bottom_pattern
;SETTING_3  = zigzag\\ntop_bottom_thickness = 0.6\\ntop_thickness = 0.8\\ntravel
;SETTING_3 _compensate_overlapping_walls_enabled = True\\ntravel_retract_before_
;SETTING_3 outer_wall = True\\nwall_thickness = 1\\nz_seam_relative = True\\nz_s
;SETTING_3 eam_type = shortest\\nz_seam_y = 30\\n\\n", "extruder_quality": ["[ge
;SETTING_3 neral]\\nversion = 2\\nname = PETG XL\\ndefinition = custom\\n\\n[met
;SETTING_3 adata]\\ntype = quality_changes\\nquality_type = normal\\nsetting_ver
;SETTING_3 sion = 4\\nextruder = fdmextruder\\n\\n[values]\\ninfill_sparse_densi
;SETTING_3 ty = 40\\n\\n"]}
````

Ghostkeeper, 2017-12-06:
This is in the g-code at the Drop Box around line 54180 (comments after the commands are mine):

    G1 F2100 X98.19 Y87.133 E265.27611 ;Infill line
    G1 X98.238 Y87.22 ;Seems to be overlap compensated infill line
    G0 F7200 X98.333 Y87.276 ;Travel connecting to inner wall
    ;TYPE:WALL-INNER
    G1 F1500 X98.18 Y87.455 E265.27963 ;Three inner wall segments
    G1 X98.017 Y87.626 E265.28317
    G1 X97.843 Y87.792 E265.28677

This is between infill and the first wall. There is no retraction going on here, so it's not applicable to this bug report.

But take a bit further in the g-code, around line 54432, where it moves to the next column:

    G1 X99.442 Y86.143 E266.1617 ;Last outer wall line segment
    G0 F1080 X99.439 Y86.151 ;Lots of coasting moves start here.
    G0 X99.338 Y86.395
    G0 X99.225 Y86.633
    G0 X99.094 Y86.874
    G0 X98.962 Y87.09
    G0 X98.813 Y87.307
    G0 X98.646 Y87.525
    G0 F7200 X98.516 Y87.677 ;Travel moves towards the infill
    G0 X97.907 Y87.157
    G1 F2700 E265.3617 ;Retract in infill area
    G0 F7200 X98.95 Y86.234 ;Travel moves to next object
    G0 X100.244 Y86.467
    G0 X103.691 Y85.346
    G0 X104.938 Y85.041
    G0 X105.019 Y83.954
    ;TYPE:FILL
    G1 F1200 E266.18565 ;Unretract in infill area of next object
    G1 F2100 X107.237 Y80.234 E266.25047 ;First infill line

As you can see, the unretract is happening only when the nozzle has moved completely to the infill area of the next object. It doesn't happen during the travel move.

Ghostkeeper, 2018-03-14:
Anyway, the original issue seems fixed to me:
[image]

People getting blobs in their prints can have 1000 reasons, but the setting can be negative, right?

closed 2018-03-14 by Ghostkeeper