Ultimaker/Cura #2668: Camera Freezes after Cura has been minimised

sctsprin, 2017-10-24:
Using Cura 3.0.3 with UM3 the camera has locked up within Cura during a print.  but the website feed works fine.

fieldOfView, 2017-10-24:
To clarify, you are talking about the (web-)camera integrated in the UM3 being displayed in Cura, not about the 3d camera in the viewport, right?

What OS do you use?

sctsprin, 2017-10-24:
Yes that’s correct
Mac OS Sierra 
It doesn’t always happen. But did today midway through a print

I still get a live image on the streaming web page though

Please excuse my brevity as I'm on my phone.

MFHFozzy, 2017-10-28:
Saw the same happen tonight with Windows 10. But at least the memory issue is gone :)

Ghostkeeper, 2017-11-06:
This lock-up occurs when Cura has lost its connection to the printer but hasn't noticed that it lost the connection. It should eventually restore itself. Does it restore?

sctsprin, 2017-11-07:
it usually does restore, but sometimes i have to wait for the job to finish and restart Cura

Ghostkeeper, 2017-11-11:
You can't restart Cura before the print job finished? Or it doesn't restore the camera image before the print job finished?

closed 2018-03-02 by ChrisTerBeke