Ultimaker/Cura #2670: Infill Pattern ignored when using 100%

shufps, 2017-10-24:
The infill pattern is ignored when using 100% infill.

That is imho a bug ... New default is concentric pattern for Top and Bottom and I don't want concentric for solid infills, because basically it stacks line on line on line on line and that can't be better than alternating +45 and -45 lines.

Please fix this :)

Thank you!

fieldOfView, 2017-10-24:
It is not a bug, it is by design. The design may need a rethink though.

A workaround is to use 99% infill, which for all intends and purposes is going to have the same result als you would expect for 100% infill.

ChrisTerBeke, 2017-10-26:
Mm for me the top and bottom infill patterns are Lines by default (UM3). Which machine and settings are you using?

mikko70, 2017-10-28:
I think then better limit  maximum 99%
All things when need big user book, is better fix before forgot it.

Ghostkeeper, 2017-11-05:
The design intent fieldOfView is speaking of is that some patterns can't be printed at 100% infill. Many infill patterns, such as grid and triangle, have lots of overlapping lines which would in theory produce 100% infill if the excess material properly flows into the gaps but in practice produce more like 60%.

We could set the infill pattern to lines instead at 100% infill. This would be more logical to an expert. However then if a user changes the infill pattern in a custom profile it will not change to lines any more, since the user has overridden that preference. So it comes down to which is more common: Changing the infill pattern to a user-specified setting or changing the skin pattern to concentric? I think changing the infill pattern is a more common use case.

Also, changing the infill pattern to lines causes the border between skin and infill to be discontinuous, since the infill typically has different line width and speed than skin. So you'll get slower speed, lower quality and less infill at that border.

closed 2018-03-02 by ChrisTerBeke