Ultimaker/Cura #2671: Feature: Filtered B-Spline Algorithm for Faster Printing

thesec, 2017-10-24:
https://www.youtube.com/watch?time_continue=4&v=6sN71fx9frk

While currently in use with firmware it might be possible to add in slicer as well?

Ghostkeeper, 2017-11-05:
I'm hitting a paywall when trying to access their paper :(

Any approximation of the vibrations of a printer will require lots of data. Measured data with quite a lot of sensors. This data will have to be provided to Cura somehow. A manufacturer may take these measurements and then provide it to Cura, but then users that don't use Cura will be left without.

Because of the effort required to take these vibration measurements, I think it's better to implement this sort of thing in Marlin. It doesn't depend on which slicer you're using then. Also, once one such Marlin implementations has been published, the others can take it over from them.

thesec, 2017-11-06:
Somewhere they claim to be in touch with Marlin already ... so lets see when that hits the repos :)

Thanks for considering this and dropping your thoughts!

closed 2018-03-26 by ianpaschal

zbowen1, 2018-05-09:
@Ghostkeeper http://s2a-lab.engin.umich.edu/publications.html Please check out our lab website - publication page!

ianpaschal, 2018-05-10:
Yeah! Go blue!

nival999, 2019-07-10:
@Ghostkeeper