Ultimaker/Cura #2672: Cura 3.0.3 under extrusion on layer change

Inglebard, 2017-10-24:
Hi,

I just notice under extrusion on Cura 3.0.3. It's visible after travel before layer change. I try to enable/disable retractation, the issue is the same. The problem was not present in Cura 2.7.

I use a Prusa MK2s with default profile "Draft Quality".

I attach an image to show the issue. (left cura 3, right cura 2).
[image]

I also join gcode generated by cura 2 and cura 3 and the STL used in the zip file.
[test-cura.zip](https://github.com/Ultimaker/Cura/files/1412392/test-cura.zip)

smartavionics, 2017-10-24:
I think the problem you have is caused by the fact that with the default settings, cura 3, does not align the z-seams on the inner and outer layers and also does not optimize the order in which the walls are printed to minimize travel. The underextrusion you are seeing is, I believe, caused by the fact that before that outer wall is printed, there is a long distance travel move and the nozzle is oozing all the while so when it gets to the outer wall, it underextrudes.
I think you can fix this by turning on the experimental wall order optimization and it may pay you to specify the z-seam position also.

Inglebard, 2017-10-25:
Hi,
Ok, so it's not a bug, it's a feature :-)

> The underextrusion you are seeing is, I believe, caused by the fact that before that outer wall is printed, there is a long distance travel move and the nozzle is oozing all the while so when it gets to the outer wall, it underextrudes.

Well, you are right. The travel move is longer on Cura 3. But does this means there is no retraction above infill ? (if it's the case, combined to non wall order optimization, it's 100% problem on a lot of prints, no ?)

> I think you can fix this by turning on the experimental wall order optimization and it may pay you to specify the z-seam position also.

Ok, I will check this ASAP.

smartavionics, 2017-10-25:
Hi
> But does this means there is no retraction above infill ?

Yes, when combing is enabled (which I think it is by default), there is no retraction above infill.

Inglebard, 2017-10-25:
Activate wall order optimization fix the issue.

closed 2017-10-25 by Inglebard

Ghostkeeper, 2017-11-05:
In 3.1 we've made the Z seam to be aligned in the inner walls as well.

Inglebard, 2017-11-06:
Ok, waiting for the next release so.

agasevangelos, 2018-02-07:
Hi!
I have the same problem on version 2.5, 2.6, 2.7!
The older version 1.5 of Cura don't have this issue.
So why is this happening?

agasevangelos, 2018-02-07:
I think I have found the solution.

Turning **Combing Mode** **off** on versions after 1.5 seems to solve the problem!
_(Which is quite weird because on version 1.5 with Combing Mode on there is not a problem.)_

bwarring24, 2018-02-10:
I have this same problem as well. Cura 15 doesn't have an issue. @agasevangelos I will try to turning that off and printing a test cube to see if it fixes it. It has been so annoying because I love having all the features in Cura 3.1 vs Cura 15, but I also want good quality prints. I'll report back tomorrow.

bwarring24, 2018-02-10:
@agasevangelos I just printed a test cube with combing off and it came out flawless! Thank you so much for figuring it out!

agasevangelos, 2018-02-10:
Hey hello!
I am glad that you found it useful and you confirmed it too!
Its a really weird problem. It took me so much time to figure it out. because on previous cura I allways had combing mode on.
I would like to hear from the cura development community how is this possible?

fyl2xp1, 2018-03-30:
I came across this issue while hunting down a slightly different problem, but it also seems to be caused by combing=on as default setting.

Please have a look at https://www.thingiverse.com/groups/prusa-i3/forums/general/topic:28926

Can this be solved by other means than disabling combing? My case is rather standard and I would expect it to work with default settings.

Ghostkeeper, 2018-04-26:
In that thread the Runtimeerror user mentions this:

> I think I'll have to find a way to tell Cura to wait for the filament coming out after a transport.

I can answer that: Try setting Retraction Extra Prime Amount to a few cubic millimetres. It extrudes a slight bit after each retraction to get the filament flowing more. It won't prevent stringing but it will prevent underextrusion after stringing if tuned correctly. Maybe that makes the lines more sturdy and thus less wobbly?

fyl2xp1, 2018-04-26:
The problem described in the thread occured due to combing which explicitely doesn't use retraction. So the Prime Amount won't help here.

Maybe faster travel speeds might help to mitigate the problem a bit.

Another compromise might be to use retraction only when crossing other structures. Currently disabling the combing mode results in a retraction hell when printing the infill pattern.

LubosD, 2018-05-22:
I'm hitting this problem on my Mk3. I've spent quite a while chasing this one down, one of the things that helped a little for the picture in the initial post was disabling "Compensate Wall Overlaps", but as I read this topic, I'm led to believe that was just a partial workaround.

I have the problem already [linked](https://www.thingiverse.com/groups/prusa-i3/forums/general/topic:28926) here. But for me also some parts of the print just don't get printed this way and I don't remember having encountering this in the past.

[image]

I was watching this print very closely and it seems the filament doesn't get primed in time. I thought something was wrong with my printer, but this issue is non-existent in Slic3r Prusa edition.

Ghostkeeper, 2018-06-29:
We should probably have a separate ticket for that issue. This one is about layer changes and it was fixed.

virdagreat147, 2023-01-07:
hey guys, i figured out a solution when i was having this issue, all you have to do is turn on "Max comb distance with no retract" and set it to like 0.001 mm, this makes it so it retracts on combing moves too, it might also help to use "extra prime amount" to compensate for the filament lost due to oozing.

enjoy! and feel free to ask questions