Ultimaker/Cura #2674: Cura's GCODE doesn't update the bed temperature if the bed temperature is set to 0

Pikamander2, 2017-10-25:
Steps to reproduce:

1. Use Cura 3.0.3 to create a GCODE file where the bed temperature is set to 0

2. On your 3D printer, manually set the bed temperature to a value higher than 0

3. Run the GCODE file on your 3D printer (in other words, begin the print)

4. The printer's bed temperature will not be set back to 0

My assumption is that Cura doesn't bother to generate the "set bed temperature" instruction if the value is set to 0. This would normally be fine since the bed temperature usually begins at 0, but in some cases you may still have the bed heat on from a prior print job or from changing your mind about what you're going to print.

smartavionics, 2017-10-25:
How about setting the bed temp to something like ambient temperate, (i.e. 20, or whatever) does that have the desired effect?

Personally, I think that a value of 0 meaning "don't alter the temperature" is just fine. After all, you may actually want to set the temperature manually before starting the print. In which case, you don't want the print to turn the bed heater off when it starts.

smartavionics, 2017-10-25:
On second thoughts, setting a low temp could lead to an over-temperature fault being triggered so that's probably not the way to go.

Pikamander2, 2017-10-25:
> Personally, I think that a value of 0 meaning "don't alter the temperature" is just fine. After all, you may actually want to set the temperature manually before starting the print.

I disagree, zero should mean zero. There should definitely be a way to leave the temperature as-is, but that should be handled with something like a blank value or a separate checkbox.

Ghostkeeper, 2017-11-05:
Currently this is intended behaviour. It says so in the setting description:

> **Build Plate Temperature**
> The temperature used for the heated build plate. If this is 0, the bed will not heat up for this print.

The description fails to say that the bed will also not cool down for this print if it is set to 0. It's not very common anyway. Putting that in the description could be confusing to novice printer operators.

A workaround for now is to set the temperature to 1 instead. We can consider separating the 0 value out to a check box, but that creates an additional setting which is also not very user friendly for novices.

Pikamander2, 2017-11-06:
I disagree with the intended behavior, but if it's going to stay that way, I think that we should make it accurate. How about one of these?

>The temperature used for the heated build plate. If set to 0, Cura will not adjust the bed temperature.

>The temperature used for the heated build plate. Set this to 0 if you want to manually heat the print bed.

sbts, 2017-11-09:
I disagree with the intended behavior as well.

tldr; Cura should ALWAYS send any enabled bed and extruder temps to the printer, regardless of the value

The suggestion @Ghostkeeper has of using 1 as a workaround isn't as far as I can tell, as any "wait for bed temperature" commands will then block until the bed temp is 1deg, I'd expect that to be equivalent to "until hell freezes over" ;-)

If cura really needs a "don't set/touch the bed temperature" then it should either be a checkbox, or a bed temp of -1.

Considering that cura may be used to slice a model that's going to be printed in some sort of food product, I'd suggest using a checkbox would make more sense, as sending 0deg or even -10 may be quite valid for some printers.

Ghostkeeper, 2017-11-15:
Maybe we need to use -274 as marker temperature 😛

Pikamander2, 2017-11-15:
That's a good point @sbts, negative values could technically be valid (maybe they are already?) as well.

I think that the best solution would be for Cura to have a checkbox that determines whether or not Cura sets the build plate temperature. This would be consistent with some of Cura's other features, such as the "Generate Support" checkbox that shows several other options if enabled.

Here's a mockup of what it could look like:

[image]

Ghostkeeper, 2017-11-24:
The setting you're proposing already exists as a machine setting. Machine settings are not visible by default, but this one happens to be exposed via the Machine Settings dialogue. You could go into the Machine Manager, then click on Machine Settings. There you can disable the heated bed entirely, which effectively prevents Cura from adding any commands for the heated bed unless you have those in your start g-code.

I've modified the description of the temperature setting a bit to be more accurate of what really happens. Good suggestion!

closed 2017-11-24 by Ghostkeeper

sbts, 2017-12-13:
@Ghostkeeper I'm not convinced using the Machine Setting "Disable Heated Bed" is the correct solution here.

I can see a need to be able to externally set the bed temp BEFORE printing commences, but at some point during the print have Cura set the temp (ie: no change for first layer, but reduced temp for subsequent layers)

Also, you may want to turn the bed off from cura as  a natural part of the job completing.

Ghostkeeper, 2018-01-01:
Then I think you should set the initial build plate temperature to match whatever you're setting in the printer. If the build plate needs to be heated before the start-up sequence of the print, then it can be added to the start of the start g-code of the printer configuration with the command `M190 S{material_bed_temperature_layer_0}`.

You can turn the bed off from Cura as a natural part of the job completing by adding that command to the end g-code.

commit 9ecfd551 (2017-11-24): Clarify description for when bed temperature is 0

Fixes #2674.