Ultimaker/Cura #2675: Closed object in spiral vase mode

cody82, 2017-10-25:
Right now it does not seam possible to print a closed object in spiral mode.
Cura prints a bottom layer but no top layer.

It should have an option to print a top layer in spiral mode.

smartavionics, 2017-10-25:
> It should have an option to print a top layer in spiral mode.

And how is it going to be able to do that? What support does the top have? You can't print onto thin air!

cody82, 2017-10-25:
Of course it would not work on all objects, but in my case it should work.
See here:
[image]
But I see your point that it is not trivial.

smartavionics, 2017-10-25:
Even in your case it's not guaranteed to work well because, remember, the top layer is a spiral and so the z-height varies around the layer so if it tried to put a roof on that it wouldn't be easy to get a nice looking result.

All, in all, I think that although it would be nice to be able to do this, the effort in making it work would rather outweigh the benefit.

If I was printing the above model, I would just use non-spiral mode, one wall and hide the z-seam on one of the corners.

ChrisTerBeke, 2017-10-25:
Vase mode is just not designed for situations like these, but rather for vases.. Pretty sure we're not gonna change anything in there.

closed 2017-10-25 by ChrisTerBeke

Ghostkeeper, 2017-11-05:
Spiralize Outer Contour doesn't mention the word "vase" though.

ChrisTerBeke, 2017-11-06:
We can change the description maybe? IMO spiralize = vase mode :P