Ultimaker/Cura #2676: Fan speeds incorrect

ChrisRiddell, 2017-10-25:
When CPE is selected the fan speed on the printer stay at 40% no matter what model you use.

I have played with the threshold and noticed no change the fan speed stays at 40% even though the settings have the 80-100 range set.

Cura 3.0.3
Windows 10
Ultimaker 2+

[image]

ChrisRiddell, 2017-10-26:
To add to this the profile on the printer it's self is set to 50% fan speed under the materials menu.

ChrisRiddell, 2017-10-28:
Update after doing some digging I'm a little confused as the gcode (ultigcode) has a M106 S204 at layer 1 that's around 80% from my under standing.

With the printers profile "custom 3" (was a modified from CPE) set to 50% it seems the printer is overriding the M106 and using tune while printing is pointless as it changes back each layer change.

So would this issue be a Firmware one?

Ghostkeeper, 2017-11-06:
The fan speed response (as in, the actual RPM of the fan blades) is far from linear. I wouldn't be surprised if it ends up at 40% if the g-code sets it to S204.

For me it gradually ramps up to S204 as well (on layer 5, since I was using UM3), so if you're seeing a fan speed of 40% it'll probably be either the firmware or the electronics/hardware causing the fan speed to be so low.

ChrisRiddell, 2017-11-07:
Only seems to be on a custom profile that has a fan speed set to 40% it overrides the gcode so would this be worth bring up in the UM2+ firmware repo?

Ghostkeeper, 2017-11-11:
I suppose so. If the g-code says S204 and the firmware does 40% instead it might be making a miscalculation on the fan speed in the firmware. Maybe something is dividing it by 2?

ChrisRiddell, 2017-11-17:
I fixed this by changing to marlin as the gcode type so something with Ulticode.

Ghostkeeper, 2017-11-26:
It seems it is. I thought it was only the temperatures and retractions, but apparently the fan is controlled in the machine too: https://github.com/Ultimaker/UM2.1-Firmware/blob/7c024fe7b762d9219caea8d1ca43c54d813a40f4/Marlin/UltiLCD2_menu_print.cpp#L434

So Cura has no influence over the fan speeds.

closed 2017-11-26 by Ghostkeeper