Ultimaker/Cura #2680: Crush Cura

DaGov, 2017-10-25:
Version: 3.0.3
Platform: Windows-7-6.1.7601-SP1
Qt: 5.8.0
PyQt: 5.8.2

Exception:
Traceback (most recent call last):
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\MainWindow.py", line 184, in _render
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\Qt\QtRenderer.py", line 138, in render
  File "E:\Program Files\Cura 3.0\plugins\LayerView\LayerPass.py", line 95, in render
    batch.render(self._scene.getActiveCamera())
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\View\RenderBatch.py", line 215, in render
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\View\RenderBatch.py", line 255, in _renderItem
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\View\GL\OpenGL.py", line 251, in createIndexBuffer
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\Mesh\MeshData.py", line 218, in getIndicesAsByteArray
MemoryError

-------------------------
Thread 0x00001b4c (most recent call first):
  File "E:\Program Files\Cura 3.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 88 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000b90 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000257c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00002700 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000144 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001f08 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000484 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\3.0\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 116 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001d30 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001494 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001498 (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001398 (most recent call first):
  File "E:\Program Files\Cura 3.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 58 in _updateThread
  File "threading.py", line 862 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000128c (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1153 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001f94 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000b0c (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00000894 (most recent call first):
  File "threading.py", line 297 in wait
  File "zeroconf.py", line 1569 in wait
  File "zeroconf.py", line 1017 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000018c0 (most recent call first):
  File "zeroconf.py", line 946 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00000f34 (most recent call first):
  File "X:\3.0\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 100 in show
  File "X:\3.0\build\inst\bin\cura_app.py", line 45 in exceptHook
  File "X:\3.0\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 686 in run
  File "X:\3.0\build\inst\bin\cura_app.py", line 82 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

Ellecross, 2017-10-26:
Hey @DaGov , what were you doing when this crash happened?

DaGov, 2017-10-26:
I sliceed 1 part for Ultimaker 3.

fieldOfView, 2017-10-26:
Please elaborate a bit. What sort of model were you slicing? Was it a model with a lot of detail or a simple model? When did the crash happen? When loading the model, during or after slicing, during or after printing?

The error indicates running out of memory. How much RAM do you have in the computer you used for slicing?

DaGov, 2017-10-26:
Model Gyro by marketing of Ultimaker 3. I saved model to flash and cura crashed.

Ghostkeeper, 2017-11-06:
Can't reproduce. I'm loading the Gyro Air model on my laptop which has 8GB of ram. Cura's memory usage climbs from 1.5GB to 2.6GB when loading up layer view, which is to be expected since the G-code itself is already 280MB and the 3D layer view tubes multiply the number of vertices significantly.

closed 2018-03-02 by ChrisTerBeke