Ultimaker/Cura #2681: open stl file generated with Solidworks 2016 unit mm

claude1000, 2017-10-25:
- Exporting the stl file from Solidworks 2016 unit in **mm** 
- Importing into Cura the file is in **Meter !** 

thanks

[image]

ChrisTerBeke, 2017-10-25:
Are you using the Solidworks plugin? If not, in Solidworks you probably have an option to set the units when exporting? Cura looks at the units described in the STL file, so if they're incorrect this is the result. If you do use it please let us know so we can check the plugin functionality.

claude1000, 2017-10-25:
I tried with / whitout
With the same issue

ChrisTerBeke, 2017-10-26:
Can you upload the STL file here so we can check the STL contents? I think the Solidworks side of the plugin uses the same exporter for STL as the normal export to STL functionality, and as Cura probably doesn't have a bug with importing STLs (the most used feature in the software) I think it's somewhere in that exporter.

fieldOfView, 2017-10-26:
> Cura looks at the units described in the STL file, so if they're incorrect this is the result.

Cura does not look at the units described in the STL file, because there is no unit described in the STL file. Cura always assumes the units are in mm when opening an STL. If SolidWorks exports in another unit, it has no way of specifying that in the STL file unfortunately.

ChrisTerBeke, 2017-10-26:
A, got confused with 3MF a bit there. Indeed STL is just vector and we assume it's sizes are in mm. This confirms my suspicion that the issue is actually in Solidworks.

fieldOfView, 2017-10-26:
Could the plugin use 3MF instead of STL to mitigate this problem?

ChrisTerBeke, 2017-10-26:
@Appesteijn something to take into account for the Solidworks plugin improvements in the future?

claude1000, 2017-10-26:
I think Solidworks export rtl files in meter ? Could you confirm ?
 I just open a ticket by SW

thanks

ChrisTerBeke, 2017-10-26:
I'm not familiar with the Solidworks source code, so can't really confirm. We have a meeting with them planned though so we'll probably bring it up.

claude1000, 2017-10-26:
Have you load my attached file in Cura 3.0.3?

ChrisTerBeke, 2017-10-26:
I don't see any attached files. That probably doesn't work when replying over email.

claude1000, 2017-10-26:
here the file

ChrisTerBeke, 2017-10-26:
>  doesn't work when replying over email

To add a file, login to GitHub and add a comment with attachment from there please.

ChrisTerBeke, 2017-10-26:
Also, we're updating the Macro to export Solidworks files instead of STL files and the Cura plugin can already read them. This will fix the export dimensions when using the Macro export. The STL export dimension issue is something that Solidworks has to fix themselves.

claude1000, 2017-10-26:
let me in touch while the magro is ready

have a nice day
claude

claude1000, 2017-10-26:
Hi
all the file attached SW , rtl , jpg

[bouton cuisine faiencerie.zip](https://github.com/Ultimaker/Cura/files/1419208/bouton.cuisine.faiencerie.zip)

Appesteijn, 2017-10-26:
The new version of the plugin will be available tomorrow. Please let us know if your problem is then solved!

claude1000, 2017-10-26:
sure

Appesteijn, 2017-10-27:
The new version is online.

claude1000, 2017-10-27:
nice
could you give me the link and plugin name

Appesteijn, 2017-10-27:
You can find it in the plugin browser in Cura

claude1000, 2017-10-27:
i guess the name is Solidworks integration ?

Appesteijn, 2017-10-27:
Yes

ChrisTerBeke, 2017-10-31:
Fixed in the solidworks integration. STL exporting has to be fixed in solidworks itself.

closed 2017-10-31 by ChrisTerBeke