Ultimaker/Cura #2683: travellers in the sliced design also laying down material

mdeans, 2017-10-26:
Tried printing a model which was hollow and had material strung across inside which seemed to correlate with the travellers in the sliced design. It was a print in one iris box design and prevented it to move when finished. The ones in dark blue are the ones that caused the issue as they ran over a couple of the moving layers
[image]

smartavionics, 2017-10-26:
Those dark blue lines are travel with no retraction. If you turn off combing, do they become light blue?

mdeans, 2017-10-26:
I didn’t have the combing setting on I don’t think. It wasn’t an option I chose to adjust. I’ll tinker with it a little and see what i can find. I’ll let you know.

Thanks for the quick response 

Regards

Mark

smartavionics, 2017-10-26:
Hi, I think that it is on by default (not 100% sure of that!)

mdeans, 2017-10-26:
Looks like combing mode off would fix it, no dark blue lines on the sliced model

Thanks!

smartavionics, 2017-10-26:
Thanks for the confirmation i already have an issue open for possibly the same bug (direct line travels that cross air that should have either have used retraction or taken another path).

fionnb, 2017-10-30:
I can confirm that sometimes v3 lays a line when traveling where it should not. It is especially problematic on the first layer, where the unwanted line often collides with the layer fill and makes the whole thing come off the bed. I can provide an example project to show this if you wish.

smartavionics, 2017-10-30:
Please do provide an example so I can investigate.

filipgoc, 2017-10-31:
Hm. I have also just discovered an intensely strange combing issue. Just posted it in Cura repo: https://github.com/Ultimaker/Cura/issues/2704
Looks like another beast, but it very well could have the same root!

Ghostkeeper, 2017-11-06:
This seems to be a duplicate of https://github.com/Ultimaker/CuraEngine/issues/625.

closed 2017-11-06 by Ghostkeeper