Ultimaker/Cura #2686: Profile changes at program startup

maukcc, 2017-10-26:
Cura V3.0.3

When one makes profile changes and not saves them as a new profile, the changes are being stored.
We would like to see a warning message, or sign, when you start the program that says you are working with a standard profile that has changed settings.
Anything that gets your attention and says : heads up, would be fine.

Or another way to go would be when you close the program, that it asks, if you would like to store your changed settings or discard them.(and then the profile would be reset to standard)

ChrisTerBeke, 2017-10-26:
Standard profiles cannot be changed. Any changes made to it are stored separately and when re-starting Cura you can click the 'reset' button in the settings fields to get back to the original values. Also Cura auto-saves these changes every 10 seconds (via the autosave plugin).

Adding a dialog to warn about saving changes might be helpful indeed.

maukcc, 2017-10-26:
One of the things that happens to us often, is when you change the infill from 20 to 60% because you need a sturdy part.
Then we start a new print the next day, and we are wondering why it takes such a long time to print, and we see it is still at 60% infill.

fieldOfView, 2017-10-26:
That's exactly what the star in the current profile is for
[image]

maukcc, 2017-10-26:
That is perfect to see **if** there are any changes.
It will not **tell** you that there are changes, unless you go looking for it.
In V3.0.3 it will not even be visible when you are viewing "Recommended" print setup

ChrisTerBeke, 2017-10-26:
v3.1 will have an indicator in recommended mode for any changes that might exist, including a 'reset to default' button for them.

maukcc, 2017-10-26:
Ah, perfect !!
Could you make it a flashing red indicator ?
Because even the black star is something you have to know and remember to look for(every single time).

The more I think about it:
people will rather have a pop up every time they save/print a file that says they have changed settings, then having to manually remember to check the star. If they do not want to see the pop up they should save to a new profile.
And on exit of the program ask if they want to make a new profile with the changes, or discard them. In any case: do not remember the changes.

ChrisTerBeke, 2017-10-26:
The indicator will be a reset button similar to the ones in front of each setting field if you change it.

About the changes dialog on quit: @Appesteijn your call :)

Ghostkeeper, 2017-11-05:
Also, the reset button doesn't work for the settings that are on the Recommended page, such as infill density. You're supposed to see the slider at 60% and know that the state of the settings is retained after restarting. It's designed to notify the user that they made changes in Custom so it's not exactly recommended any more.

Another use case is that a user always wants to print at 10% infill and hates that it gets reset to 20% every time (personally I think that 20% is a bit much and you're better off with an extra top skin layer). Or you just love gradual infill. I don't think that resetting the user settings at every start is desired.

maukcc, 2017-11-06:
Standard operating procedure at shutdown of any program is the question:

>  Are you sure you want to close without saving your changes?

Then if a user always wants to print at 10% infill, they know they have to make a new profile. And Cura will start with the last used profile at next startup. No problem, no worries.

Another thing to consider is legal issues.
The machine vendors put a lot of time and effort in making good profiles for their machines. So when users start altering the settings and prints get wonky, the user can not blame the vendor for making poor settings. 
Maybe it is even a good idea to have a line in the quality files that says:
"user_profile_changes_allowed": true (or false)

In any case, just remembering changes without notification at program end is a poor (legal) decision.

nallath, 2017-11-08:
I don't think the legal notification is required. If you drive 250 km/h with a car, you also don't need a legal notification that stuff might not last as long if you do this.

closed 2018-03-02 by ChrisTerBeke