Ultimaker/Cura #2687: systray icon issues (X11)

cmock, 2017-10-26:
Since Cura 3, there is a systray icon that has two problems:

- It does nothing, neither responds to any mouse button nor displays any information
- in a multiple desktop setting, it appears all desktops except the one Cura is running in

I could see such an icon providing a quick glance at print progress/status, but the way it is, it's just a waste of screen real estate. May I ask you at least make it configurable whether it appears?

Setup: Debian stable, tint2 as the taskbar, running Cura 3.0.3 AppImage.

[image]

Ghostkeeper, 2017-11-06:
Yeah, I've seen that as well. It's caused by Cura now giving OS-notifications ("toast") if the window is minimised. This notifies you if your slice is finished, the print has finished, etc. This functionality requires a tray icon, at least on Windows (I don't know about X11). Qt creates the icon on all platforms.

Any idea what we could put in the menu of the tray icon?

cmock, 2017-11-11:
The "minimized" part would explain why it is only on the other desktops.

The only notifications I've seen so far are about an upload to octoprint being finished, which I don't care about in my workflow.

What I would like is a progress bar when a print is running (or on any long-running operation), but optimally that'd be in the icon so I don't have to hover the mouse over it or click.

And an option to disable it would also be nice.

Ghostkeeper, 2017-11-19:
Anything that pops up a blue message at the bottom of the screen should also pop up a toast from the system tray.

I agree that a progress bar would be very useful. And maybe clicking the icon should un-minimise the Cura window.

Ghostkeeper, 2018-02-05:
Our project manager just removed the feature request from our planning because it's been untouched for 12 weeks. Or rather, because it would be untouched for 12 weeks next week... (?)

fieldOfView, 2018-02-05:
Ha! There’s activity on the issue now, so you’re stuck with it another 12 weeks!

jake-nz, 2022-02-27:
Especially for those of us using Cura as a slicer only, there's no need to system tray messages. Perhaps a setting to opt-out of both messages and the icon would be a good idea? It would definitely be fine for me.

fieldOfView, 2022-02-28:
I have made a set of PRs to add a preference for the system tray icon (defaulting to true, as per the current behavior):
https://github.com/Ultimaker/Cura/pull/11573
https://github.com/Ultimaker/Uranium/pull/788

closed 2022-09-08 by nallath