Ultimaker/Cura #2688: Cura 3.0.3 Octoprint Connections disappear when Cura closes

tmanstls, 2017-10-26:
Having and issue where my Octorpint instances disappear if I close and reopen Cura. If I leave it open, they have no issue, but once it gets closed the next time they are gone. I have 3 printers setup each with their own Octoprint server and they all disappear after closing and I have to re-add.

[curalog.zip](https://github.com/Ultimaker/Cura/files/1418757/curalog.zip)

fieldOfView, 2017-10-26:
Could you attach cura.cfg (zipped) from the same directory you got cura.log from? I think I found the error, but would like to be sure.

tmanstls, 2017-10-26:
[Curacfg.zip](https://github.com/Ultimaker/Cura/files/1418863/Curacfg.zip)

fieldOfView, 2017-10-26:
I have fixed cura.cfg for you. Please let me know if the OctoPrintPlugin behaves now, and I can make it more resilient for these issues.
[cura_cfg.zip](https://github.com/Ultimaker/Cura/files/1419009/cura_cfg.zip)

tmanstls, 2017-10-26:
It is fixed. Thank You!
Can I ask what the problem was?

fieldOfView, 2017-10-26:
Your cura.cfg had remnants of an older version of the octoprintplugin, which lacked some data. The plugin does not handle missing that data gracefully. I'll fix it so this does not happen to other users.

fieldOfView, 2017-10-27:
Fixed here: https://github.com/fieldOfView/OctoPrintPlugin/commit/e60272dee1329485772f1c627cd0260f5dec9ece
Issue can be closed

closed 2017-10-27 by nallath